RIOT-OS/RIOT #3201: ng_net: Odd length packet snips cause invalid check sum

authmillenon, 2015-06-14:
If an odd length packet snip occurs in a packet and is not the last snip (in the order the packet is supposed to be, not in the list's order) in a packet it will generate a wrong check sum by way the check sum is calculated: If a snip is of odd length a 0 is padded to the last bit, since it groups the bytes to 2 bytes. This de-facto includes a non-existing 0-byte into the middle of the packet. While in most cases this does not present a problem, it is problematic if the application uses odd-length headers (as e.g. ZEP in its current version).

PeterKietzmann, 2015-06-30:
@authmillenon could you provide a simple patch or something to test this? Is this mybe a Hack'n'ACK candidate?

authmillenon, 2015-06-30:
This is a little bit more complex to be just "a simple patch". You probably have to "merge" two or more snips temporarily to get this fixed.

authmillenon, 2015-06-30:
What I can do is add a test application that show-cases this bug...

authmillenon, 2015-06-30:
But not before the Hack'n'ACK itself.

PeterKietzmann, 2015-06-30:
Yes, that's what I meant. And no stress, I just wanted to trigger this discussion...

authmillenon, 2015-06-30:
Will add a (for now deactivated) test to #3248.

authmillenon, 2015-07-01:
Added test to #3296 (https://github.com/RIOT-OS/RIOT/pull/3296/files#diff-8dceb44f59303bdb12036fd100abbab2R68 but as stated currently deactivated at https://github.com/RIOT-OS/RIOT/pull/3296/files#diff-8dceb44f59303bdb12036fd100abbab2R266)

OlegHahm, 2015-08-17:
Anything here?

authmillenon, 2015-08-17:
Nope, but as long as we don't use any weird odd-length header formats between payload and IPv6 we should be fine.

OlegHahm, 2015-08-18:
You mean between IPv6 header and its payload? What about the original problem with ZEP?

authmillenon, 2015-08-18:
No I mean the IPv6 and the highest layer's payload in its payload. The problem occurs upper to IP when a packet snip that does not represent the payload of the packet is of odd numbered length. So if e.g. the transport layer would have a header length of 9. In ZEP this was fixed by just creating a new UDP payload snip and replacing it with the snipified payload.

authmillenon, 2015-12-03:
Fixed by #4343

closed 2015-12-03 by authmillenon