RIOT-OS/RIOT #3221: How to generate random numbers?

sgso, 2015-06-18:
I'm working on libcoap, which needs some randomized value here and there. Nothing crypto-critical but I'd like to do it the right way and the existing choices/docs are not really helpful. And this might become more important with things like DTLS in the future.

**What's the recommended way to generate random numbers on RIOT?**
- I can check for the existence of RANDOM_NUMOF and if positive use periph/random (but see below)
- Only a few architectures implement periph/random. Do I fall back to stdlib rand()?
- But I can't really call srand() because any other module, that uses rand() might call srand() and then there's seed chaos. Is that harmful?

**I'm also working on a hardware platform that provides a hardware pseudo random number generator:**
- The periph/random interface documentations states:
 
> In case of a hardware random number generator, this peripheral is initialized and powered on. 
> If such a device is not present, it depends on the implementation how a source for randomness is created and initialized.

- So this is the place to support a hardware PRNG even if it's just a deterministic pseudo generator, right?
-  I'm advised to initialize the platform PRNG with at least 80 Bytes of entropy, but this has to suceed once somebody decides to call `void random_init(void)`. If I happen to find an entropy source worth 80 Bytes (I doubt it) that's fine but there's no way to let the caller know if that was not the case. But if you try to get entropy from some source dependent on external events, there might just not be enough of it available yet but maybe 10 seconds later when network activity picked up.
- Shouldn't entropy gathering and PRNG initialization be decoupled?
- Then an interface like `genrand_init_by_array(uint32_t init_key[], int key_length)` from the mersenne twister would be much better. It offloads the duty of entropy gathering to the caller. If there's a TRNG available you pass NULL.
- My specific PRNG allows to add further entropy to the existing one. I'd like to support that. 
- What's the purpose of the `Random` module? Is it meant to be permanently and tightly coupled to the mersenne twister implementation? Then it should better be called the MersenneTwister module. Or was it once supposed to be the thing I feel is missing right now?
- Brings me to: Are there any plans to optionally gather some entropy centrally? For example from timing differences between IPC messages or the scheduler? It would only be an active codepath until enough entropy is available...
- Shouldn't there be one API that can be configured at compile time to be backed by either a hardware TRNG, hardware PRNG, software PRNG (mersenne), a custom drop in, libc callback, etc...?

authmillenon, 2015-06-19:
> What's the recommended way to generate random numbers on RIOT?

I normally use genrand_uint32() of `sys/include/random.h` for the NDP stuff. The hardware random interface was thought to be a truly "randomness" source to set the seed for this (dangling ADCs, timer, cpu_id etc.) afaik, but I don't know if this was ever done.

emmanuelsearch, 2015-06-19:
so, linked with https://github.com/RIOT-OS/RIOT/pull/2963 ?

kaspar030, 2015-06-19:
Not really. This ask for a standard way of getting random numbers, #2963 would be a possible user for such an API.

sgso, 2015-06-20:
@authmillenon: So including NDP "wastes" about 2500 Bytes of RAM just for random numbers if there's an hardware generator (true or pseudo) available? That's a low hanging fruit ;)

authmillenon, 2015-07-24:
@sgso it's no 80 bytes of entropy (this can be changed easily), but see https://github.com/RIOT-OS/RIOT/pull/3425

ArnoRabenstein, 2015-10-07:
For a rng in hardware, there is Infineon's XMC1100 (ARM CORTEX M0) with pseudo random number generator (PRNG) on chip.

sgso, 2015-10-07:
@ArnoRabenstein No. It's a pseudo RNG which needs a seed of 80Bit (not Bytes of course like I stated above). That's what my question was about.

closed 2019-09-10 by stale[bot]