RIOT-OS/RIOT #3242: core: simplify the reboot interface?!

haukepetersen, 2015-06-23:
#3219 made me look at the kernel's reboot interface once more. It seems to me, that it can be simplified to some extend. Here is my proposal:

- remove the `mode` parameter to the reboot function, as it is not used and I don't see any usecase for this
- remove the reboot function from core entirely and just use the `reboot_arch()` in the first place
- add some additional implementation for `reboot_arch()`, that explicitly states that we expect this function never to return (i.e. by implementing it using some kind of loop, etc.).

Any opinions on this?

gebart, 2015-06-23:
I like this proposal.

+1

haukepetersen, 2015-06-23:
alright, then I will prepare a small PR for this.

OlegHahm, 2015-06-23:
1. I think the parameter was originally introduced for POSIX or BSD compatibility reasons. 
2. Calling `*_arch()` from outside the kernel feels wrong for me.

haukepetersen, 2015-06-23:
lets continue this discussion in the PR I created: #3246

cgundogan, 2015-10-27:
@haukepetersen since #3246 got closed, is there still a need to have this issue open?

haukepetersen, 2015-10-27:
nope

closed 2015-10-27 by haukepetersen