RIOT-OS/RIOT #3249: `ng_ndp` calls `fib_get_next_hop()` with a `NULL` parameter

cgundogan, 2015-06-24:
Please refer to [ng_ndp.c#L419](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/network_layer/ng_ndp/ng_ndp.c#L419) and https://github.com/RIOT-OS/RIOT/pull/3233#issuecomment-114654397

I opened this issue in order to keep better track of the discussion.
@authmillenon could you  briefly explain the surrounding code of the line being in question? What is intended to happen there?

authmillenon, 2015-06-24:
From @BytesGalore at https://github.com/RIOT-OS/RIOT/pull/3233#issuecomment-114723987

> @authmillenon I'm against exposing the internally used FIB pointers to the "outside".
After such call, the caller could never be certain if the pointer is still valid and connected to the desired address bytes.

Currently IPv6 is the only the only user of `fib_get_next_hop()` and due to the nature of the scheduler it is pretty much guaranteed that it uses the address without interruption (all it does is: get address from FIB, look it up in NC, if available take l2 address, if not copy address into NC). 

>  Additionally this would result in bypassing the read/write protection of the internal FIB structures.

I'm not convinced by that argument. We deal with constraint systems here, which means, if we can save some bytes of RAM and ROM vs. strict modularisation, we should at least think about, if the strict modularisation makes sense. Since we work on most systems with no RAM protection at all, I would argue that the read/write protection can be bypassed anyway (from a security stand-point). From a usage stand-point you can also document such things that the address is NOT to be modified outside the context of the FIB (see e.g. my exposure of the IPv6 PID in `ng_ipv6.h`).

authmillenon, 2015-06-24:
@cgundogan what exactly do you want me to explain? It is just more convenient (memory-wise and usage-wise) to use a pointer for this lookup, since it would make every assignment there a `memcpy()` and every `NULL` pointer check a `ng_ipv6_addr_is_unspecified()` check. I'd like to prevent that. The only compromise I came up with that would work with the current prototype is

```diff
diff --git a/sys/net/network_layer/ng_ndp/ng_ndp.c b/sys/net/network_layer/ng_ndp/ng_ndp.c
index 3ee288f..9c1bc54 100644
--- a/sys/net/network_layer/ng_ndp/ng_ndp.c
+++ b/sys/net/network_layer/ng_ndp/ng_ndp.c
@@ -415,13 +415,18 @@ kernel_pid_t ng_ndp_next_hop_l2addr(uint8_t *l2addr, uint8_t *l2addr_len,
 #ifdef MODULE_FIB
     size_t next_hop_size;
     uint32_t next_hop_flags = 0;
+    ng_ipv6_addr_t next_hop_actua;
+
     if ((next_hop_ip == NULL) &&
-        ((fib_get_next_hop(&iface, (uint8_t *)next_hop_ip, &next_hop_size,
+        ((fib_get_next_hop(&iface, next_hop_actual.u8, &next_hop_size,
                           &next_hop_flags, (uint8_t *)dst,
                           sizeof(ng_ipv6_addr_t), 0) < 0) ||
                           (next_hop_size != sizeof(ng_ipv6_addr_t)))) {
         next_hop_ip = NULL;
     }
+    else {
+        next_hop_ip = (ng_ipv6_addr_t *)&next_hop_actual;
+    }
 #endif
 
     if ((next_hop_ip == NULL)) {            /* no route to host */
```

but that is really clunky IMHO.

authmillenon, 2015-06-24:
@BytesGalore do I need to initialize `next_hop_size` with some value here, btw?

BytesGalore, 2015-06-24:
@authmillenon just a quick response for your last question, it depends ;) .
Basically it expects the caller to provide the available size for the address bytes (shame on me I didn't wrote this in the doxy, sorry for that).
Then it overwrites the value with the actual required size for the address.

So only if you provide a sufficient large `next_hop_size`, the address bytes are copied, else only the actual required size is overwritten in `next_hop_size` to indicate the caller how much "space" you would require for the address.

cgundogan, 2015-06-25:
https://github.com/RIOT-OS/RIOT/pull/3250#issuecomment-115216907
>authmillenon commented 6 hours ago
We still need a solution for usage in IPv6 though.

How will ipv6 react if #3250 gets merged? I mean, does this still fit into the logic of ipv6? I don't really understand the line of code btw. What is your intend by putting a `NULL` param into `fib_get_next_hop()`?

cgundogan, 2015-06-25:
BTW I have no idea why I only encounter this crash when I merge #3233

authmillenon, 2015-06-25:
IPv6 will react as such, that routing won't work... We will need at least my proposed fix for that. Or something better. :) What code are you referring to? The current implementation is erroneous.

PR #3272 (2015-06-30): ng_ndp: fix ng_ndp_get_next_hop() for FIB fix in #3250
Fixes #3249
closed 2015-06-30 by cgundogan