RIOT-OS/RIOT #3252: gnrc: improve unittest coverage

OlegHahm, 2015-06-24:
This a list of modules that are part of the gnrc network stack and there state concerning unittest:

* [ ] Netdev Ethernet

 *No Unittests*
* [ ] Network Interfaces

 **19 unittests exist**
 - [ ] Generic Network Interface Header

     *No Unittests*
* [ ] Network Protocol Registry

 **12 unittests exist**
* [ ] Packet Buffer

 **38 unittests exist**
* [ ] NETAPI
 *No Unittests*
 - [ ] NETAPI Test Framework
  
   *No Unittests*
* [ ] Nomac

 *No Unittests*
* [ ] SLIP

 *No Unittests*
* [ ] NDP

 *No Unittests*
* [ ] 6LoWPAN

 *No Unittests*
 - [ ] 6LoWPAN Contexts

   *No Unittests*
 - [ ] 6LoWPAN Fragmentation

   *No Unittests*
 - [ ] 6LoWPAN Header Compression

   *No Unittests*
 - [ ] 6LoWPAN Network Interfaces

   *No Unittests*
* [ ] IPv6

 *No Unittests*
 - [ ] IPv6 Address

    **67 unittest exist**
    - [ ] IPv6 Header

    **17 unittest exits**
    - [ ] IPv6 Extension Header

    *No Unittests*
    - [ ] IPv6 Extension Routing Header

   *No Unittests*
 - [ ] IPv6 Neighbor Cache

   **27 unittest exist**
 - [ ] IPv6 Network Interfaces

   **34 unitttest exist**
* [ ] ICMPv6

   *No Unittests*
 - [ ] ECHO

   *No Unittests*
* [ ] Upper Layer/Inet Checksum

 **5 unittest exist**
* [ ] RPL

   *No Unittests*
 - [ ] RPL Source Routing Header

   *No Unittests*
* [ ] UDP

   *No Unittests*
* [ ] NHDP

   *No Unittests*

The intention of this issue is to keep track of the coverage of unittests per module and improve it.

OlegHahm, 2015-06-24:
Please note: currently I only counted the number of the existing unittests without stating anything about their quality or coverage. I will update this information ASAP.

authmillenon, 2015-06-24:
At least for the `netapi` part of UDP there is #3248. There is still need for `ng_udp_hdr_build()` and such.

authmillenon, 2015-06-24:
Some unittests based on `pexpect` or similar modules for the network device drivers are also thinkable. We can use the driver's test apps as a starting point.

PeterKietzmann, 2015-06-24:
@A-Paul may be interested in this list.

OlegHahm, 2015-06-24:
@authmillenon, feel free to edit the list and add links to PRs etc. We can also change the format if someone has a better idea. So far, it's more or less a dump.

A-Paul, 2015-06-28:
@PeterKietzmann, yes he is. Thanks. :-)

By now (still no familiar with the structure and history of RIOT) I'm confused by figuring out:
1. How the list entries map to the header files in *sys/include/net*.
2. Where to find the specific untittests for the new network stack in *"tests/unittests/"* and other directories under *"tests/"* (Unsure if *"tests/udp"* is the only exception).
3. And again, how those map to the modules and/or header files.

@OlegHahm or @authmillenon can you give me some advice in this?

PeterKietzmann, 2015-06-29:
1. @A-Paul, currently it's a bit confusing as we have both the old and the new network stack in master. Modules related to the new stack are prefixed with `ng_`. In *sys/include/net* there are only header files related to the new stack. Except the device driver interfaces you should find all relevant headers there. For example: "Netdev Ethernet" relates to `ng_netdev_eth.h`, "Packet Buffer` relates to `ng_pktbuf.h`. .. Which module do you have in mind to work on?
2. Unittests are placed in *tests/unittests/**. The tests placed in *tests/** are more to understand as demo of a module.
3. For example: As stated above, the module "IPv6 Address" already has some unit tests. You can find them in *RIOT/tests/unittests/tests-ipv6_addr/*. If you feel insecure if a tests relates to the new network stack, you can have a look inside the Makefile.include. There you can see that a `ng_`-prefixed module is included.

Did I answer your questions?

authmillenon, 2015-06-29:
> 2. Unittests are placed in tests/unittests/. The tests placed in *tests/* are more to understand as demo of a module.

That is not true, IMHO. `tests/` contains what it says: test application. Most of them utilize the shell and are missing a proper execution script, but they are still tests. Anyway my rational behind creating a new test application for every netapi integration test was, that `tests/unittests` is to be used only for tests not involving threads, timers, IPC etc. (because those things are hard to map in simple unittests) and threads/IPC related tests to use seperate test applications. The unittests are already quite big and don't fit in total on all of our supported boards, so adding thread-stacks to it the situation wouldn't get better.

> 3. For example: As stated above, the module "IPv6 Address" already has some unit tests. You can find them in RIOT/tests/unittests/tests-ipv6_addr/. If you feel insecure if a tests relates to the new [...]

Maybe it is helpful, if someone™ with some insight into the network stack can update the list with the according header files and test locations ;-)

PeterKietzmann, 2015-06-29:
@authmillenon naming them as "demo" was a bit imprecise, I agree. I was thinking about driver tests I guess. Anyway, in this issue we are talking about the bare unittest, right? This should be the location that we are focusing on. IMO that's something where "not-so-__gnrc__-experienced" people can support.

authmillenon, 2015-06-29:
> Anyway, in this issue we are talking about the bare unittest, right? This should be the location that we are focusing on. IMO that's something where "not-so-gnrc-experienced" people can support.

@PeterKietzmann Yes, but they are the least interesting in terms of complexity for the stack. The netapi calls are really what glues everything together/where the most seems to go wrong ;-).

OlegHahm, 2015-06-29:
> @authmillenon naming them as "demo" was a bit imprecise, I agree. I was thinking about driver tests I guess. 

@A-Paul was referring to these tests as [Smoke Tests](https://en.wikipedia.org/wiki/Smoke_testing_%28software%29)

PeterKietzmann, 2015-06-29:
@authmillenon I never wanted to say that they are not interesting :-) ! @OlegHahm I'm not sure I got what you wanted to tell me. Anyway, it would be nice if you manage this testing, I'll better keep me out to avoid inconsistencies. @katezilla could also be interested

OlegHahm, 2015-06-29:
@PeterKietzmann, I just think that "smoke testing" seems to be the most correct term for most of the applications located directly under `tests/`.

authmillenon, 2015-06-30:
Can someone shed some light on how the coverage is defined?

OlegHahm, 2015-06-30:
In theory, see https://en.wikipedia.org/wiki/Code_coverage. In practice, as I wrote, this is for now just a very rough estimation based on the number of functions per interface and the number and range of parameters.

OlegHahm, 2015-07-02:
@authmillenon, am I right to assume that network interface PIDs are not checked when added to IPv6 or 6LoWPAN (e.g. in `ng_sixlowpan_netif_add()`)? If yes, we should document this behavior.

authmillenon, 2015-07-02:
What do you mean with "checked"? Usually they are called by `ng_netif_add` so there is no need to check if they are an interface.

OlegHahm, 2015-07-02:
What happens if a I call, for instance, `ng_sixlowpan_netif_add()` with a random ID? Will it check if this ID is a valid PID or just create the 6lowpan interface?

authmillenon, 2015-07-02:
Will add it.

A-Paul, 2015-07-06:
While looking through the existing tests I've seen a lot of locallly defined data stubs. I may be usefull if those who are sufficient for other test could go into into `unittests-constants.h`. Ideally including a short comment which special state/case they represent.

OlegHahm, 2015-07-06:
Seems like a valid proposal. Maybe you can prepare a PR for this?

miri64, 2019-01-16:
Where do we go with this? On one hand, we are still striding towards a better test coverage, on the other, there is #10187. Finally, I don't think the list above is up-to-date ;-).

closed 2019-09-10 by stale[bot]