RIOT-OS/RIOT #3255: vtimer: `vtimer_msg_receive_timeout()` does not timeout

authmillenon, 2015-06-24:
In #3248 I discovered, that `vtimer_msg_receive_timeout()` does not timeout. Investigating this further it seems, that `vtimer_callback_msg()` actually gets called, but the main thread never receives the message in `vtimer_msg_receive_timeout()` for some reason.

gebart, 2015-06-29:
is this related to #3269 ?

authmillenon, 2015-06-29:
Close: I didn't initialize any packet queue for this particular thread (`main`) so the message had no way to be delivered. So this isn't a bug, but was just wrong usage.

closed 2015-06-29 by authmillenon

OlegHahm, 2015-06-29:
Maybe the documentation can be improved!?

authmillenon, 2015-06-29:
Of what?

* of `vtimer_msg_receive_timeout()` does not really apply, since `msg_receive()` is also "affected"
* of `msg_receive()`, maybe. But than I would in general state, since you never know if a message comes from ISR or not.
* of `msg_init_queue()`?
* of `main()` (is there even documentation on that?)

OlegHahm, 2015-06-30:
At least the documentation of `vtimer_msg_receive_timeout()` should state that a message queue is required, because the messages are always sent non blocking here.

> But than I would in general state

In general state...what?

authmillenon, 2015-06-30:
*state*, that a message queue is required.

OlegHahm, 2015-06-30:
Required? No, one has just to be aware that messages might get lost if you may receive messages from ISR (or other non-blocking senders).