RIOT-OS/RIOT #3256: Setting constants on compile time doesn't really set them everywhere

Lotterleben, 2015-06-24:
During the work on https://github.com/RIOT-OS/RIOT/pull/3022, @BytesGalore noted that adding the possibility to set constants on compile time the way we do it currently doesn't set the constant everywhere (I'll just qute him):

>@Lotterleben changed to configure on compile time, but this may produce unwanted side-effects.
	The defines are set right for the FIB/Universal address only (bad news).  

>When including the `ng_fib.h` in a main, for instance in the `hello-world` example	 (with `USEMODULE += fib` of course) and calling say `make UNIVERSAL_ADDRESS_SIZE=17`, 
	the main will use the `UNIVERSAL_ADDRESS_SIZE` default value of **`16`**, while the FIB will initialize and use it with the new value **`17`**.  (same applies for all other defines)

>You can reproduce it by e.g. printing the value of the defines in the FIB (e.g. in `fib_init()`) and in the main accordingly.  
	So I recommend to not provide this compile time configuration. 
	
This is done in at least one other instance, namely the RPL Makefile (https://github.com/RIOT-OS/RIOT/blob/9ecaea4038d401659f79e6ee93b82692c90c2372/sys/net/routing/rpl/Makefile).

BytesGalore, 2015-06-24:
Small addition, when you set `QUIET = 0` you can track which objects are compiled with the provided command line compile time constant.

OlegHahm, 2015-06-24:
`CFLAGS="-DUNIVERSAL_ADDRESS_SIZE=17" make`  will work.

BytesGalore, 2015-06-24:
@OlegHahm yup, but the intention was to provide this configurability when just using the FIB module and passing the according parameter to `make` for the project that uses the FIB.
In that case it depends on when the specific Makefile of a module is used for compiling the .o/.a files.
If I add this line to say `RIOT/sys/net/network_layer/fib/Makefile`, its only parsed and used when compiling this module.
All other ones that includes the headers before this Makefile has been processed, uses the default defined values from the included headers.
I observed this behaviour in #3022, but now I removed the commit to get this merged.
But here is the part for `UNIVERSAL_ADDRESS_SIZE` I had in the Makefile:
```
# set the universal address size
ifneq (,$(UNIVERSAL_ADDRESS_SIZE))
	CFLAGS += -DUNIVERSAL_ADDRESS_SIZE=$(UNIVERSAL_ADDRESS_SIZE)
endif
include $(RIOTBASE)/Makefile.base
```
Of course I enhanced the `ng_fib.h` with:
```
#ifndef UNIVERSAL_ADDRESS_SIZE
#   define UNIVERSAL_ADDRESS_SIZE (16)
#endif
```
accordingly :) (sorry for the wall)

OlegHahm, 2015-06-25:
Ah, okay, sorry for my lazyness to read through the referenced PR. With the mentioned snip for the (application's?) Makefile: didn't this work, either?

Lotterleben, 2015-06-28:
You mean of  #3022? It was for a while because I'd asked @BytesGalore for this functionality, but then he discovered the issue described and then it wasn't part of the PR anymore.

OlegHahm, 2015-06-28:
Yes, I was talking abouit #3022, but I couldn't find
```
# set the universal address size
ifneq (,$(UNIVERSAL_ADDRESS_SIZE))
    CFLAGS += -DUNIVERSAL_ADDRESS_SIZE=$(UNIVERSAL_ADDRESS_SIZE)
endif
include $(RIOTBASE)/Makefile.base
```
there.

Lotterleben, 2015-06-28:
Thats because Martin already rebased...

OlegHahm, 2015-06-28:
Because, it didn't work?

Lotterleben, 2015-06-28:
Exactly. We opened the issue with a bit of a delay there :/

cgundogan, 2015-08-20:
I also faced the same issue with the former rpl implementation. It appears that adding the `CFLAGS` addition to the local `Makefile` only affects the specific module's compilation.
I guess what should work is when you put your modification into `Makefile.base`, because this is included by all other modules. This is however definitely the wrong way to go.

PeterKietzmann, 2017-11-27:
@Lotterleben is this still present? @cladmi can you give an estimation about the current status?

cladmi, 2017-11-28:
If I understand correctly what is described here:

In a module `Makefile`, lets say `sys/net/super_module` there is
```
MODULE = super_module

ifneq(,$(ENABLE_SUPER_POWER))
  CFLAGS += -DSUPER_POWER=$(ENABLE_SUPER_POWER)
endif

include $(RIOTBASE)/Makefile.base
```

In a **header** for this module there is
```
#ifndef SUPER_POWER
#define SUPER_POWER (0)
#endif
int enable_something_cool = SUPER_POWER;
```
And running `make ENABLE_SUPER_POWER=1` only affects `super_module` and not every modules.

---

If I understood correctly, I say this is intended behavior and is even a feature.

Changing CFLAGS in a module `Makefile` only affects it.
To affect its API == something done in its headers, it should be done in a `Makefile.include`.

For example:
* in `sys/Makefile.include` for `gnrc_ipv6` and the `SOCK_HAS_IPV6` configuration https://github.com/RIOT-OS/RIOT/blob/2017.10/sys/Makefile.include#L14 (same for embunit)
* in the module Makefile.include which is included by `sys/Makefile.include` or another like for newlib https://github.com/RIOT-OS/RIOT/blob/2017.10/sys/Makefile.include#L64 (its called `newlib.mk` but its the same idea)

The global configuration is handled by:
* environment variables/command line definitions
* exported variables in the application Makefile (some are always exported https://github.com/RIOT-OS/RIOT/blob/master/makefiles/vars.inc.mk)
* files included directly or indirectly by the global `Makefile.include`.

All what is done in an module Makefile only affects the given module, (or also module build through the DIRS variable if you export variables but I have not seen this in the current repository).

---

So I think this could be closed with an "work as intended" by documenting it somewhere.

cladmi, 2017-11-30:
I will write a PR to document "For developers, how to make a module configurable".

miri64, 2020-06-30:
@leandrolanzieri @jia200x @akshaim is this still a problem with Kconfig?

maribu, 2022-09-16:
Let me close this and see if anyone complains :)

closed 2022-09-16 by maribu