RIOT-OS/RIOT #3258: gnrc/ng_ipv6_addr: potential overrunning array?

PeterKietzmann, 2015-06-24:
I don't know much about the implementation. So just understand this as a question and close the issue if it's stupid. Assuming in [`ng_ipv6_addr_init_prefix`](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/network_layer/ng_ipv6/addr/ng_ipv6_addr.c#L60) the variable `bits` becomes `128`. Then `bytes = bits/8` will become `16`. In [line 76](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/network_layer/ng_ipv6/addr/ng_ipv6_addr.c#L76) and following the variable `bytes` acts as an index for the field which is initialized as `uint8_t u8[16]`

cgundogan, 2015-06-24:
`if (bits % 8)` comes to rescue here, the code in question will not be executed if bits is 128

closed 2015-06-24 by cgundogan

PeterKietzmann, 2015-06-24:
Sure. I guess I need to go home...