RIOT-OS/RIOT #3261: gnrc/ng_netdev_eth: potential dereferencing nullpointer

PeterKietzmann, 2015-06-25:
In ng_netdev_eth the line  [`if (payload)`](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/link_layer/ng_netdev_eth/ng_netdev_eth.c#L376) indicates that `payload` might be `NULL`. Some lines below [`_addr_set_multicast(hdr->dst, payload)`](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/link_layer/ng_netdev_eth/ng_netdev_eth.c#L399) is called. Inside this function, `payload` is [dereferenced](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/link_layer/ng_netdev_eth/ng_netdev_eth.c#L342). What if `payload` **is** actually `NULL`? I don't know if this is a realistic situation or not. Just close this issue if it's superfluous!

authmillenon, 2015-06-25:
Is this a duplicate of https://github.com/RIOT-OS/RIOT/pull/3238? Can you please refer to the lines you are talking of?

authmillenon, 2015-06-25:
This IS a duplicate of #3238.

PeterKietzmann, 2015-06-25:
Yes it is was a duplicate. PS: I linked the regarding code lines

OlegHahm, 2015-08-18:
With #3238 being closed without merging, I think this is still unfixed.

PR #3651 (2015-08-18): gnrc: null pointer checks
fixes #3261, alternative to #3238 and fixed version of #3473.
closed 2015-08-18 by authmillenon