RIOT-OS/RIOT #3275: Hard Fault on iot-lab_m3 platform

rousselk, 2015-06-30:
Hello everyone,

When performing extensive experimentations of my RIOT code on IoT-LAB M3 nodes, I regularily encounter `HARD FAULT` panics.

Reading the ARM Cortex-M manual indicates that Hard Faults are non recoverable, hardware-generated exceptions, usually provoked by interruption (mis)management.

Since I also get the impression that in my experimentations, I am sometimes facing problems that might indicate that some IRQs I should receive do not "register" and launch the corresponding handler, I wonder if there could be some problem in RIOT's interruption management on that platform.
More precisely: the ARM NVIC is a complex beast, and there are many ways to have subtle bugs related to its configuration.

Is there many experiments/projects on the `iot-lab_m3` platform? Is anybody encountering similar problems?

OlegHahm, 2015-06-30:
@rousselk, I recently did some experimentation of lower layer stuff on the IoT-LAB_M3 nodes and cannot recall to encounter any hard faults there, but experiments were mostly running only for a couple of minutes. Can you provide some kind of showcase to reproduces the failures?

rousselk, 2015-07-01:
Hello @OlegHahm,

I encounter these problem while working on my PhD thesis stuff, which has diverged quite a lot from RIOT's master branch... My code is in my own repositories (`rousselk/RIOT` branch `s-cosens`, with test applications stored in `roussel/projects` branch `my-work`).

I can imagine, though, that trying to work on these code bases would be much hassle for you.

These hard faults are not frequent. To be honest, they aren't much of a problem to me _per se_, but the problem of sometimes-unresponsive IRQs is: it is now blocking me, at a moment where I have almost no more time left to finish experiments...

After reading the Cortex-M manual about hard faults and their origin, I am now wondering if the two problems could be linked--i.e.: due to the same cause.
I have looked for hard fault management on Cortex-M3, and found these Web pages:
* http://www.freertos.org/Debugging-Hard-Faults-On-Cortex-M-Microcontrollers.html;
* http://www.freertos.org/RTOS-Cortex-M3-M4.html which insists on interrupts misconfiguration as the main cause for hard faults (and other bugs)

... So it seems that the Cortex NVIC is a complex beast, hard to configure correctly, even for FreeRTOS developers, who have quite some experience on the subject; it seems possible to me, then, that these kind of problems are now biting us...

gebart, 2015-07-01:
Can you run your code with the debugger attached and show us the backtrace
when you get your hard fault?

Hard faults are typically the application code dereferencing an invalid
pointer and reading/writing from/to an invalid memory location (or a
disabled peripheral), or unhandled interrupts.

Check that all vectors in you interrupt vector table point to valid
functions.

OlegHahm, 2015-12-04:
@rousselk, have you tried with latest RIOT master code?

A-Paul, 2015-12-16:
Hi @rousselk I'm in the process of selecting open issues for the 2015.12 release notes. Can you please confirm if described failures are still happening reproducible?

cgundogan, 2016-01-05:
I am in favor of closing this issue. The codebase that @rousselk used while encountering the described problem is quite old and AFAIK there is no known issue for `iotlab-m3` boards that is similar to the one stated here.
As long as this issue is not reproducible (and thus unconfirmed) I see no reason to mention this in the release notes.

For now, I will move this issue to the next milestone.

rousselk, 2016-01-06:
Hello,

I was effectively working on an old codebase (derived from release 2014.12) when encountering these hard faults. I have not worked much with the new codebase, but neither Shuguo Zhuo (who is replacing me since my PhD is ending) nor I did encounter any hard faults since now.

I think that issue can be discarded and closed.

rousselk, 2016-01-06:
Hello,

I was effectively working on an old codebase (derived from release 2014.12) when encountering these hard faults. I have not worked much with the new codebase, but neither Shuguo Zhuo (who is replacing me since my PhD is ending) nor I did encounter any hard faults since now.

I think that issue can be discarded and closed. If something similar happens again, another issue can then be reopened.

closed 2016-01-06 by rousselk

cgundogan, 2016-01-06:
@rousselk great, thanks for the information