RIOT-OS/RIOT #3280: make: buildtest target always generates some native artifacts

phiros, 2015-06-30:
When you execute a buildtest for a MCU group which is not ```qemu-i386``` a directory ```native```
gets created in the bin folder with an empty archive file in it. I am not really sure if you can call this a bug but I find it odd (as a user I would call this unexpected behavior). 
I only found out about this because the ```bin/native``` directory doesn't seem to be created as part of the buildtest for loop. Therefore, #3268 does not get rid of this directory after the buildtest is done.

BTW: I wasn't sure what label to pick. I think it might be useful to create something like an ```low-priority-issue``` or even a label called ```unexpected behavior``` for such problems.

Example:
```
/home/.../examples/hello-world$ ls
Makefile  README.md  main.c
/home/.../examples/hello-world$ BUILDTEST_MCU_GROUP=msp430 make buildtest
Building for chronos .. success
Building for msb-430 .. success
Building for msb-430h .. success
Building for telosb .. success
Building for wsn430-v1_3b .. success
Building for wsn430-v1_4 .. success
Building for z1 .. success
/home/.../examples/hello-world$ ls -l bin/
total 32
drwxrwxr-x 12 philipp philipp 4096 Jun 30 20:44 chronos
drwxrwxr-x 11 philipp philipp 4096 Jun 30 20:44 msb-430
drwxrwxr-x 11 philipp philipp 4096 Jun 30 20:44 msb-430h
drwxrwxr-x  2 philipp philipp 4096 Jun 30 20:44 native
drwxrwxr-x 11 philipp philipp 4096 Jun 30 20:44 telosb
drwxrwxr-x 11 philipp philipp 4096 Jun 30 20:44 wsn430-v1_3b
drwxrwxr-x 11 philipp philipp 4096 Jun 30 20:44 wsn430-v1_4
drwxrwxr-x 11 philipp philipp 4096 Jun 30 20:44 z1
/home/.../examples/hello-world$ ls -l bin/native/
total 4
-rw-rw-r-- 1 philipp philipp 8 Jun 30 20:44 empty-archive.a

```

gebart, 2015-07-01:
It comes from the ar-wrapper test on this line: 
https://github.com/RIOT-OS/RIOT/blob/master/Makefile.include#L134

It creates the empty archive in native if that is your default target, if you `export BOARD=somethingelse` it will be placed there instead.

An `rm` at the end of the shell command line should be sufficient:
```diff
-ifneq (0, $(shell mkdir -p $(BINDIR); $(AR) rc $(BINDIR)empty-archive.a 2> /dev/null; echo $$?))
+ifneq (0, $(shell mkdir -p $(BINDIR); $(AR) rc $(BINDIR)empty-archive.a 2> /dev/null; echo $$?; rm -f $(BINDIR)empty-archive.a))
```

phiros, 2015-07-02:
fixed by #3302

closed 2015-07-02 by phiros