RIOT-OS/RIOT #3284: mpu: Use memory protection unit (MPU) for protecting memory accesses

gebart, 2015-07-01:
Many Cortex M3 and Cortex M4 processors have memory protection units that can be configured to disallow access to parts of the memory address space from different processes. We are currently not using any MPU features in RIOT.

Some starting ideas:

 - A basic implementation: protect the ISR/Supervisor mode stack from accesses from application code.
 - More advanced: per-thread access control, e.g. preventing applications from accessing hardware peripherals which they should not be using.

The main purpose of implementing MPU support would be to help in detecting software buffer overflows during testing, but I guess it would add application security/robustness if implemented correctly.

Processor hardware support:

ARM MPU:
 - Atmel SAM3X8E
 - STM32F4, STM32F3, some STM32F1's
 - CC2538

Other MPU:
 - Kinetis K60 (Freescale MPU)
 - Kinetis KW2x (Freescale MPU)

OlegHahm, 2015-07-01:
@StefanPfeiffer, you had a bachelor student working on this some time ago. Can you share some information about the outcome somehow?

kaspar030, 2015-07-01:
Pebble OS uses MPU to provide a safe application environment. They run their apps in an MPU protected memory area, and provide access to system functionality through privilegue checked SVC system calls.

We could take a more detailed look...

grundrauschen, 2015-07-13:
Hello all together,

Because of holidays and moving back from Sweden I did not have time to answer earlier.
I am the student mentioned by @OlegHahm and wrote a Bachelor Thesis about that topic, which I handed in march 2014.
As a disclaimer I have to mention, that the thesis is based on a branch created by @StefanPfeiffer already in summer 2013 with some additions added later.
Because I was busy with other work in between, I haven't followed the progress of RIOT in the meantime and the kernel might look completely different nowadays.
I added the link to my thesis in [1], but because it is written in german and I am not sure how many of you are able to read it, I will summarise it a bit.

I used the STM32F4Discovery board as a platform for my development and based it on the mentioned branch by @StefanPfeiffer. Because it was a thesis which viewed on the possibility and way how to use the MPU I concentrated on implementing a solution for the STM32F4 and F3 without creating a proper API for other architectures.

In general I rewrote major parts of the kernel to be able to implement the protection.
The first part is a memory management function. The kernel was at time of writing not able to allocate memory areas and the used stacks for the threads were assigned manually in the source code.
The provided functions opened the possibility to get memory areas of a certain size (matching the block sizes of the MPU) for new threads.
Based on these I created functions to generate new thread, which called for a new memory area and set the properties of the thread with the connected rights and memory areas.
Furthermore I modified the scheduler so that it not just switches the process, but additionally reprogramms the MPU setting and give the thread only write access to its threads but neither to management areas (Thread controll block) nor global variables.
The kernel itself has full access to the memory.

Do to this architecture the threads are running in a sandbox and violations are registered by the kernel and the processes are restarted a certain amount of times until they are finally shut down due to misbehaviour.

To be able to communicate with other threads and the kernel itself by this sandboxed threads, I implemented Supervisor Calls to send a message to other threads by handing it over to the kernel, which sends it to the receiving process. The message itself is saved in the designated memory areas of the threads.

Additionally to this implementation I created some tests, which shows for some common memory problems, that the protection is sound.

In case the Riot-project wants to use parts of my bachelor thesis, it would of course be necessary to have a look on how the kernel changed in the meantime and adjust the code to that.
Further more it would be necessary to provide a stable API to also work with other processors.
Never the less I used just CMSIS functions and common ARM Cortex M assembler calls, as far as I remember, so that an extension to other ARM Cortex M controllers should be not that complicated.

If you have more questions concerning my thesis, please just ask.

Sincerely,
Tobias

[1]: http://tobias.spline.de/bachelorarbeit.pdf

Venemo, 2016-03-16:
@grundrauschen Out of curiosity, where can we take a look at your code?

authmillenon, 2016-05-07:
An intersting read on this topic (especially in the context of per-thread-MP) [the paper "On μ-Kernel construction" by Liedtke](http://www.di.ens.fr/~pouzet/cours/systeme/bib/publ_1995_liedtke_ukernel-construction.pdf) might be an interesting read.

OlegHahm, 2016-05-31:
Since I read in the other PR that @hexluthor is working on something like this, I reassign.

hexluthor, 2016-06-21:
See my PR #5564. It doesn't protect any supervisor / hardware memory from application code as suggested by @gebart, but it does implement a low-level Cortex-M MPU driver and use it to protect against stack overflows.

grundrauschen, 2016-07-24:
Sorry for writing back so late, but I overread the messages.

closed 2019-09-10 by stale[bot]