RIOT-OS/RIOT #3301: fib_tests.test_fib_15_get_lifetime exp -1 was 1

gebart, 2015-07-02:
I get this failure consistently on the first boot after a flash (on `mulle`), but not on any subsequent resets after the first boot.

I have no idea where to look for the source of the error.

Below is a terminal dump (I pressed the hardware reset button a couple of times after the first to verify consistency)

```
kernel_init(): This is RIOT! (Version: 2014.12-2055-g5788)
kernel_init(): jumping into first task...
......................................................................................................................................
.....................................................................................
fib_tests.test_fib_15_get_lifetime (.../riot/tests/unittests/tests-fib/tests-fib.c 557) exp -1 was 1
......................................................................................................................................
............................................................................................
run 445 failures 1
kernel_init(): This is RIOT! (Version: 2014.12-2055-g5788)
kernel_init(): jumping into first task...
......................................................................................................................................
......................................................................................................................................
......................................................................................................................................
...........................................
OK (445 tests)
kernel_init(): This is RIOT! (Version: 2014.12-2055-g5788)
kernel_init(): jumping into first task...
......................................................................................................................................
......................................................................................................................................
......................................................................................................................................
...........................................
OK (445 tests)
kernel_init(): This is RIOT! (Version: 2014.12-2055-g5788)
kernel_init(): jumping into first task...
......................................................................................................................................
......................................................................................................................................
......................................................................................................................................
...........................................
OK (445 tests)
kernel_init(): This is RIOT! (Version: 2014.12-2055-g5788)
kernel_init(): jumping into first task...
......................................................................................................................................
......................................................................................................................................
......................................................................................................................................
...........................................
OK (445 tests)
kernel_init(): This is RIOT! (Version: 2014.12-2055-g5788)
kernel_init(): jumping into first task...
......................................................................................................................................
......................................................................................................................................
......................................................................................................................................
...........................................
OK (445 tests)
```

BytesGalore, 2015-07-02:
@gebart the failing test on your first boot determines if a read timepoint is lower than the current time + about one second.
It seems that this is not the case.

I guess after the flashing the current time is not cleared or just garbage, which produces a wraparound in the current time +1s and results that the left passed value in `timex_cmp()` is way larger.

BytesGalore, 2016-03-22:
@gebart do you still experience this issue?

gebart, 2016-03-22:
I will be away from my office until next week, I'll test this then.

BytesGalore, 2016-07-26:
@gebart does this issue still apply?

gebart, 2016-07-26:
nope, does not occur anymore

closed 2016-07-26 by gebart