RIOT-OS/RIOT #3303: ng_sixlowpan: API documentation is imprecise

OlegHahm, 2015-07-02:
1. Documentation for `ng_sixlowpan_init()` does not specify what happens if called more than once.
2. Documentation for `ng_sixlowpan_init()` says that `EOVERFLOW` is only returned, if two many threads are running. Isn't rather the return value of `thread_create()` returned instead? (I don't want to look this up myself for black box testing.)

authmillenon, 2015-07-02:
1. Will do
2. I might have just copied the return documentation from `thread_create()`, since they *are* directly piped through. So I'm not 100% sure what the issue is you point out here.

OlegHahm, 2015-07-02:
> I might have just copied the return documentation from thread_create(), since they are directly piped through. So I'm not 100% sure what the issue is you point out here. 

My question is basically: will I get this error only with too many threads running or also if, e.g., no memory for the sixlowpan thread is available.

authmillenon, 2015-07-02:
No for this case 1 would apply.

OlegHahm, 2015-07-02:
Means: I will get this error only with too many threads running?

authmillenon, 2015-07-02:
Yes.

PR #3306 (2015-07-03): ng_sixlowpan: clarify documentation
Fixes #3303.
closed 2015-07-03 by OlegHahm