RIOT-OS/RIOT #3304: make: buildtest 'eats' compiler warnings and errors

phiros, 2015-07-02:
A look into ```Makefile.buildtest``` reveals that we suppress all compiler warnings and errors (except the return value) in the buildtest target (the target which is executed by ```compile_test.py``` for CI test runs):
``` 
LOG=$$(env -i \
                    [...]
                    $(MAKE) -j$(NPROC) 2>&1 >/dev/null) ; \
```
As an additional result of this output redirection no details are given to the user/developer in case of a failed build. 

I have two questions:

1. Do we really want to suppress all compiler warnings in the buildtest?
2. Shouldn't we display at least the error messages generated in the buildtest in some form?

Regarding 2: I think it is feasible to modify the ```compile_test.py``` and the ```buildtest``` target so that we could collect the error messages and display them at the end of the generated output of the buildtest in a summary kind of style (after the skipped / success / failed summary). 

Regarding 1: Just suppressing warnings won't give us an incentive to fix these issues and I am not sure whether most devs are even aware of all warnings. 
BTW: I realize that some of those warning should actually be ignored but I'd say those are in the minority.

For reference a small selection of warnings you __could__ see (without the output redirect) on current master when you run the ```buildtest``` target for ```examples/hello-world```: 

```
Building for chronos .. /home/philipp/RIOT/cpu/cc430/cc110x_cc430.c: In function 'cc110x_write_reg':
/home/philipp/RIOT/cpu/cc430/cc110x_cc430.c:111:27: warning: variable 'i' set but not used [-Wunused-but-set-variable]
/home/philipp/RIOT/cpu/cc430/flashrom.c:55:16: warning: 'prepare' defined but not used [-Wunused-function]
/home/philipp/RIOT/cpu/cc430/flashrom.c:63:6: warning: 'finish' defined but not used [-Wunused-function]
```

```
Building for openmote .. /home/philipp/RIOT/cpu/cc2538/periph/gpio.c: In function 'isr_gpiob':
/home/philipp/RIOT/cpu/cc2538/periph/gpio.c:551:50: warning: array subscript is above array bounds [-Warray-bounds]
             state = gpio_config + reverse_pin_lut[GPIO_PXX_TO_NUM(PORT_B, bit)];
                                                  ^
/home/philipp/RIOT/cpu/cc2538/periph/gpio.c: In function 'isr_gpioc':
/home/philipp/RIOT/cpu/cc2538/periph/gpio.c:576:50: warning: array subscript is above array bounds [-Warray-bounds]
             state = gpio_config + reverse_pin_lut[GPIO_PXX_TO_NUM(PORT_C, bit)];
                                                  ^
/home/philipp/RIOT/cpu/cc2538/periph/gpio.c: In function 'isr_gpiod':
/home/philipp/RIOT/cpu/cc2538/periph/gpio.c:601:50: warning: array subscript is above array bounds [-Warray-bounds]
             state = gpio_config + reverse_pin_lut[GPIO_PXX_TO_NUM(PORT_D, bit)];
                                                  ^
success
```

```
Building for redbee-econotag .. /home/philipp/RIOT/cpu/arm7_common/syscalls.c: In function '_kill':
/home/philipp/RIOT/cpu/arm7_common/syscalls.c:238:15: warning: unused parameter 'pid' [-Wunused-parameter]
 int _kill(int pid, int sig)
               ^
/home/philipp/RIOT/cpu/arm7_common/syscalls.c:238:24: warning: unused parameter 'sig' [-Wunused-parameter]
 int _kill(int pid, int sig)
                        ^
In file included from /home/philipp/RIOT/core/include/clist.h:22:0,
                 from /home/philipp/RIOT/core/include/tcb.h:25,
                 from /home/philipp/RIOT/core/include/sched.h:85,
                 from /home/philipp/RIOT/core/sched.c:24:
/home/philipp/RIOT/core/sched.c: In function 'sched_run':
/home/philipp/RIOT/core/include/kernel_macros.h:53:14: warning: cast increases required alignment of target type [-Wcast-align]
             ((TYPE *) ((char *) __m____ - offsetof(TYPE, MEMBER))); \
              ^
/home/philipp/RIOT/core/include/clist.h:41:49: note: in expansion of macro 'container_of'
 #define clist_get_container(NODE, TYPE, MEMBER) container_of(NODE, TYPE, MEMBER)
                                                 ^
/home/philipp/RIOT/core/sched.c:68:26: note: in expansion of macro 'clist_get_container'
     tcb_t *next_thread = clist_get_container(sched_runqueues[nextrq], tcb_t, rq_entry);
                          ^
/home/philipp/RIOT/core/thread.c: In function 'thread_measure_stack_free':
/home/philipp/RIOT/core/thread.c:109:25: warning: cast increases required alignment of target type [-Wcast-align]
     uintptr_t *stackp = (uintptr_t *)stack;
                         ^
/home/philipp/RIOT/core/thread.c: In function 'thread_create':
/home/philipp/RIOT/core/thread.c:149:17: warning: cast increases required alignment of target type [-Wcast-align]
     tcb_t *cb = (tcb_t *) (stack + stacksize);
                 ^
/home/philipp/RIOT/core/thread.c:154:31: warning: cast increases required alignment of target type [-Wcast-align]
         uintptr_t *stackmax = (uintptr_t *) (stack + stacksize);
                               ^
/home/philipp/RIOT/core/thread.c:155:29: warning: cast increases required alignment of target type [-Wcast-align]
         uintptr_t *stackp = (uintptr_t *) stack;
                             ^
/home/philipp/RIOT/core/thread.c:164:10: warning: cast increases required alignment of target type [-Wcast-align]
         *(uintptr_t *) stack = (uintptr_t) stack;
          ^
success
```

OlegHahm, 2015-07-02:
> Do we really want to suppress all compiler warnings in the buildtest?

Definitely not!

gebart, 2015-07-02:
See also #1121

gebart, 2015-07-02:
What is the limit on variable contents length in shell scripts like this? I imagine the message log can become quite long for some platforms.

authmillenon, 2015-07-02:
You can always pipe to a temporary file.

phiros, 2015-07-02:
Well, as far as I know most modern shells are capable of storing considerably more than just a few kB in variables (supposedly as much as they can malloc).
However, it might still be safer to go for temporary files instead (like @authmillenon suggested).

gebart, 2015-07-03:
@phiros could you open a PR enabling stderr messages in `make buildtest`?

phiros, 2015-07-04:
I already did (#3314) :-) However, this doesn't help that much since ```compile_test.py``` does some weird stuff with stdout and stderr. I will look into this soon.

phiros, 2015-07-27:
Solved by #3403

closed 2015-07-27 by phiros