RIOT-OS/RIOT #3315: tests/driver_k2xrf overflows ram on nucleo-f334

phiros, 2015-07-04:
This could either be a broken toolchain (unlikely; observed this both on ubuntu 15.04 and 14.04) or something that our current build script does not want to tell us (see #3304). 

Error log:
```
/tests/driver_kw2xrf$ LANG=C BOARD=nucleo-f334 make clean all
Building application "driver_kw2xrf" for "nucleo-f334" with MCU "stm32f3".

"make" -C /home/philipp/RIOT/boards/nucleo-f334
"make" -C /home/philipp/RIOT/core
"make" -C /home/philipp/RIOT/cpu/stm32f3
"make" -C /home/philipp/RIOT/cpu/cortexm_common
"make" -C /home/philipp/RIOT/cpu/stm32f3/periph
"make" -C /home/philipp/RIOT/drivers
"make" -C /home/philipp/RIOT/drivers/kw2xrf
"make" -C /home/philipp/RIOT/sys
"make" -C /home/philipp/RIOT/sys/auto_init
"make" -C /home/philipp/RIOT/sys/auto_init/netif
"make" -C /home/philipp/RIOT/sys/compat/hwtimer
"make" -C /home/philipp/RIOT/sys/net/crosslayer/ng_netapi
"make" -C /home/philipp/RIOT/sys/net/crosslayer/ng_netif
"make" -C /home/philipp/RIOT/sys/net/crosslayer/ng_netif/hdr
"make" -C /home/philipp/RIOT/sys/net/crosslayer/ng_netreg
"make" -C /home/philipp/RIOT/sys/net/crosslayer/ng_pktbuf
"make" -C /home/philipp/RIOT/sys/net/crosslayer/ng_pktdump
"make" -C /home/philipp/RIOT/sys/net/link_layer/ng_nomac
"make" -C /home/philipp/RIOT/sys/newlib
"make" -C /home/philipp/RIOT/sys/od
"make" -C /home/philipp/RIOT/sys/posix
"make" -C /home/philipp/RIOT/sys/ps
"make" -C /home/philipp/RIOT/sys/shell
"make" -C /home/philipp/RIOT/sys/shell/commands
"make" -C /home/philipp/RIOT/sys/timex
"make" -C /home/philipp/RIOT/sys/uart0
"make" -C /home/philipp/RIOT/sys/vtimer
/usr/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: /home/philipp/RIOT/tests/driver_kw2xrf/bin/nucleo-f334/driver_kw2xrf.elf section `.bss' will not fit in region `ram'
/usr/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: region `ram' overflowed by 752 bytes
collect2: error: ld returned 1 exit status
make: *** [all] Error 1
```

gebart, 2015-07-04:
Examine the map file to see which functions or files are using the large
amounts of ram.

jremmert-phytec-iot, 2015-07-05:
The Board nucleo-f334 is not blacklisted in the kw2xrf-driver yet as this is done in ```tests/driver_at86rf2xx```.

So far, the defined size os 16k RAM apparently was sufficient. #3310 reordered the RAM configuration to 12k RAM and 4k CCMRAM. I'm not really familiar of the change in #3310 and the difference of CCMRAM and RAM. But shouldn't the overall size (RAM+CCMRAM=16k) also be used after the partition?

Any objections in blackisting the nucleo-f334 Board in the kw2xrf-test app as in the at86rf2xx driver? Would open a PR then.

kaspar030, 2015-07-05:
On 07/05/15 18:49, Jonas R. wrote:
> Any objections in blackisting the nucleo-f334 Board in the kw2xrf-test
> app as in the at86rf2xx driver? Would open a PR then.
No. If this test ever worked before on the f334, it was pure luck.

jremmert-phytec-iot, 2015-07-05:
Opened #3318

PeterKietzmann, 2015-07-06:
As #3318 was merged I think we can close this issue.

closed 2015-07-06 by PeterKietzmann