RIOT-OS/RIOT #3324: cpu/native: LPM, should we add a break in switch case?

PeterKietzmann, 2015-07-07:
Please compare [this line](https://github.com/RIOT-OS/RIOT/blob/master/cpu/native/lpm_cpu.c#L132). Even if it's an `exit` call, I wonder if we should add a `break;` there for the sake of completeness.

PeterKietzmann, 2015-07-07:
@LudwigOrtmann any suggestions?

kaspar030, 2015-07-07:
Wouldn't that just be one unnecessary instruction?

PeterKietzmann, 2015-07-07:
Maybe yes. But that is what I'd like to figure out. Feel free to close this issue if this is (really) the case

A-Paul, 2015-07-07:
As long as there exists **no** condition in `real_exit()` that could make it *return*, `break` would be considered "dead code". Code analyzers may give a warning on that sequence.

PeterKietzmann, 2015-07-07:
Ok, then let's close this issue

closed 2015-07-07 by PeterKietzmann