RIOT-OS/RIOT #3338: ng_pktbuf: strange debug output

daniel-k, 2015-07-08:
I'm not sure if it's worth an issue, but I open it anyway. While digging through `_pktbuf_add_unsafe(...)` ( [link](https://github.com/RIOT-OS/RIOT/blob/1e262e3e73b5b077e42e0b71fab758bd944070b0/sys/net/crosslayer/ng_pktbuf/ng_pktbuf.c#L197) ),  I stumpled upon some strange debug outputs.

```c
new_pktsnip = (ng_pktsnip_t *)_pktbuf_internal_alloc(sizeof(ng_pktsnip_t));
DEBUG("pktbuf: allocated (new_pktsnip = %p) ", (void *)pkt);
```
I guess it should print `(void *)new_pktsnip` instead. `pkt` might even be `NULL`.

```c
new_pktsnip->data = _pktbuf_internal_alloc(size);
DEBUG("pktbuf: allocated (new_pktsnip->data = %p) ", pkt->data);
```
The same as above, but in this case additionally, dereferencing `pkt` would lead to a hardfault if `NULL` (see [link](https://github.com/RIOT-OS/RIOT/blob/1e262e3e73b5b077e42e0b71fab758bd944070b0/sys/net/crosslayer/ng_pktbuf/ng_pktbuf.c#L212) )

Maybe I didn't understand your intention for this output correctly, but anyway I really had to struggle to understand what this function is doing exactly to debug my memory leak (this happens quite fast).

Please see this as a call for a better documentation. As I know have understood how the whole `ng_pktbuf` is supposed to work (at least I think so :P) I'd offer my help for this task.

authmillenon, 2015-07-09:
You are right, these debug outputs are wrong, will fix.

> Maybe I didn't understand your intention for this output correctly, but anyway I really had to struggle to understand what this function is doing exactly to debug my memory leak (this happens quite [...]

If you see a lack of documentation you do understand, please go ahead and rework it. If you don't ask and lets try to figure out a better documentation together. In general I should point out however, that @haukepetersen and I are planing to rework the packet buffer API slightly.

authmillenon, 2015-07-09:
(Maybe its better if you open a new issue for the last point since this is unrelated to this issue).

PR #3342 (2015-07-09): ng_pktbuf: fix debug output for _pktbuf_add_unsafe
Fixes #3338.
closed 2015-07-09 by PeterKietzmann