RIOT-OS/RIOT #3341: ng_nativenet crashes when hammered v2

kaspar030, 2015-07-09:
(this is a different problem than #3222)

using "ping6 -f %tapX" from two threads on linux to riot at the same time, I can reliably crash native:

```
(gdb) run tap0
Starting program: /home/kaspar/src/riot.7/examples/ng_networking/bin/native/ng_networking.elf tap0
RIOT native uart0 initialized.
RIOT native interrupts/signals initialized.
LED_GREEN_OFF
LED_RED_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

kernel_init(): This is RIOT! (Version: 2014.12-2147-gb227)
kernel_init(): jumping into first task...
UART0 thread started.
uart0_init() [OK]
RIOT network stack example application
All up, running the shell now
> 
Program received signal SIGSEGV, Segmentation fault.
0x0804f1a4 in _find (prev_ptr=0x8079fe4 <_nomac_stack+7460>, node_ptr=0x8079fe8 <_nomac_stack+7464>, ptr=0x8076bec <_buf+588>)
    at /home/kaspar/src/riot.7/sys/net/crosslayer/ng_pktbuf/_pktbuf_static.c:140
140	            node = (_used_t *)node->next;
(gdb)
```

authmillenon, 2015-07-09:
Did you check this with #3316 merged? Because this fixed a lot of these problems.

kaspar030, 2015-07-09:
yes, this is on master.

PeterKietzmann, 2015-07-09:
@kaspar030 what are the prerequisites for such a ping from linux to tapX? Just doing it throws "unknown host". I didn’t investigate further.

kaspar030, 2015-07-09:
@PeterKietzmann on the linux side, "ip link set tapX up", then "ping6 -f <riot-ll-address>%tapX"

kaspar030, 2015-07-09:
@PeterKietzmann I usually create a persistent tap using "openvpn --mktun --dev tap0". That way you can RIOT as non-root, and the tap survives RIOT restarts.

kaspar030, 2015-07-09:
Can you also reproduce this?

PeterKietzmann, 2015-07-10:
I just gave it a quick try yesterday but I wasn't able to set the tap device up, I guess. Still the "unknown host" thing. I assume I need to spend a bit time the get this running.

OlegHahm, 2015-07-10:
I can reproduce.

kaspar030, 2015-07-10:
Alright. To rule out nativenet, we need a l2 test application.
Could someone write a shell command that takes "<if-pid> <target-mac-addr> <num> [<delay>] [<extra-data>]" as arguments,
and sends <num> l2-packets with ethertype 0xffff to <target-mac-addr> via <if-pid> with <delay> us delay between packets.
The payload of the packet should contain the nr of the packet, starting with 0 for the first. If <extra-data> is given, the packet should be filled with that number of garbage bytes (e.g., zeroes).

authmillenon, 2015-07-10:
> @PeterKietzmann I usually create a persistent tap using "openvpn --mktun --dev tap0". That way you can RIOT as non-root, and the tap survives RIOT restarts.

The new and shiny way to go (with iproute2) would be

```bash
sudo ip tuntap add dev tap0 mode tap user ${USER}
sudo ip link set tap0 up
```

This is how it is set up in the `tapsetup.sh` script (and the unified script introduced in #3182) minus addition to a bridge. This also creates a persistent tap interface accessible by the user `USER`.

authmillenon, 2015-07-10:
About the shell command: will do: wanted to do something similar on ICMPv6 echo requests anyways. But how does this compare to sending pings from Linux to RIOT? I guess the combination of receiving and sending to two distinct sources is what destroys RIOT here.

authmillenon, 2015-07-10:
something like `l2ping` would be far more comparable I think. Sending those from the Linux-side can be done with `scapy`.

kaspar030, 2015-07-10:
> But how does this compare to sending pings from Linux to RIOT?

Just that will test *only* gnrc l2 receive. If that doesn't break when
used from two RIOT senders to one receiver, fine. Our problem is in a
higher level.

Next step is to write a simple responder that just bounces back the
packet unchanged. If that doesn't break, but icmpv6 does, the problem is
ipv6 or icmpv6.

kaspar030, 2015-07-10:
> something like l2ping would be far more comparable I think. Sending those from the Linux-side can be done with scapy.

Keep it simple to rule out errors in the tests... Sending l2 in a loop should not be more than +5 lines to _netif_send().

authmillenon, 2015-07-13:
One remark:

```bash
ping6 -f %tap0
```

does not work for me, however

```bash
ping6 -f -I tap0 <ipv6 address of RIOT node>
```

is working great. @PeterKietzmann maybe that is helpful for you.

On further note: I don't just get SIGSEGVs here:

```
Program received signal SIGBUS, Bus error.
__memcpy_ia32 () at ../sysdeps/i386/i686/multiarch/../memcpy.S:94
94	../sysdeps/i386/i686/multiarch/../memcpy.S: No such file or directory.
(gdb) where
#0  __memcpy_ia32 () at ../sysdeps/i386/i686/multiarch/../memcpy.S:94
#1  0x0804ddd7 in ng_netif_hdr_set_src_addr (hdr=0x80788a4 <_buf+36>, 
    addr=0x8078266 <recv_buffer+6> "\202\373(\332\332p\206\335`", addr_len=6 '\006')
    at /home/martine/Repositories/RIOT-OS/RIOT/sys/include/net/ng_netif/hdr.h:145
#2  0x0804e968 in _rx_event (netdev=0x8077c40 <ng_netdev_eth>)
    at /home/martine/Repositories/RIOT-OS/RIOT/sys/net/link_layer/ng_netdev_eth/ng_netdev_eth.c:515
#3  0x0804e850 in dev_eth_rx_handler (dev=0x8077c20 <dev_eth_tap>)
    at /home/martine/Repositories/RIOT-OS/RIOT/sys/net/link_layer/ng_netdev_eth/ng_netdev_eth.c:481
#4  0x0804d45c in _isr (ethdev=0x8077c20 <dev_eth_tap>)
    at /home/martine/Repositories/RIOT-OS/RIOT/cpu/native/ng_net/dev_eth_tap.c:80
#5  0x0804e4bc in _isr_event (dev=0x8077c40 <ng_netdev_eth>, event_type=1)
    at /home/martine/Repositories/RIOT-OS/RIOT/sys/net/link_layer/ng_netdev_eth/ng_netdev_eth.c:344
#6  0x08057f69 in _nomac_thread (args=0x8077c40 <ng_netdev_eth>)
    at /home/martine/Repositories/RIOT-OS/RIOT/sys/net/link_layer/ng_nomac/ng_nomac.c:88
#7  0x555e8c6b in makecontext () at ../sysdeps/unix/sysv/linux/i386/makecontext.S:87
#8  0x08077c40 in dev_eth_tap ()
#9  0x08075980 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

This one I already know from flooding RIOT to RIOT:
```
Program received signal SIGTRAP, Trace/breakpoint trap.
0x00000000 in ?? ()
(gdb) where
#0  0x00000000 in ?? ()
#1  0x0804bfd8 in isr_cpu_switch_context_exit () at /home/martine/Repositories/RIOT-OS/RIOT/cpu/native/native_cpu.c:153
#2  0x0807c00c in _nomac_stack ()
#3  0xc290af00 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

```
Program received signal SIGSEGV, Segmentation fault.
0x08073269 in __isr_stack ()
(gdb) where
#0  0x08073269 in __isr_stack ()
#1  0x555e8bcb in setcontext () at ../sysdeps/unix/sysv/linux/i386/setcontext.S:39
#2  0x0804bfd8 in isr_cpu_switch_context_exit () at /home/martine/Repositories/RIOT-OS/RIOT/cpu/native/native_cpu.c:153
#3  0x0804c0aa in cpu_switch_context_exit () at /home/martine/Repositories/RIOT-OS/RIOT/cpu/native/native_cpu.c:181
#4  0x0804b443 in native_irq_handler () at /home/martine/Repositories/RIOT-OS/RIOT/cpu/native/irq_cpu.c:277
#5  0x555e8c6b in makecontext () at ../sysdeps/unix/sysv/linux/i386/makecontext.S:87
#6  0x00000000 in ?? ()
```

```
Program received signal SIGABRT, Aborted.
0x55577425 in __kernel_vsyscall ()
(gdb) where
#0  0x55577425 in __kernel_vsyscall ()
#1  0x555d6607 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#2  0x555d9a33 in __GI_abort () at abort.c:89
#3  0x55610e53 in __libc_message (do_abort=do_abort@entry=1, fmt=fmt@entry=0x5570b633 "*** %s ***: %s terminated\n")
    at ../sysdeps/posix/libc_fatal.c:175
#4  0x556a385b in __GI___fortify_fail (msg=<optimized out>, msg@entry=0x5570b61b "stack smashing detected")
    at fortify_fail.c:37
#5  0x556a37ea in __stack_chk_fail () at stack_chk_fail.c:28
#6  0x0804c742 in make_message (format=0x807c184 <_nomac_stack+8164> "\253\001\002", 
    argp=0x8073248 <__isr_stack+8040> "h2\a\b") at /home/martine/Repositories/RIOT-OS/RIOT/cpu/native/syscalls.c:253
#7  0x0807c184 in _nomac_stack ()
#8  0x080616b0 in ?? ()
#9  0x0807c174 in _nomac_stack ()
#10 0x0807bf9c in _nomac_stack ()
#11 0x00000000 in ?? ()
```

Looks more like there is some stack corruption happening.

authmillenon, 2015-07-13:
```
(gdb) call ps()
enableIRQ + _native_in_isr
dev_eth_isr: possibly lost interrupt.
	pid | name                 | state    Q | pri | stack ( used) | location   
	  1 | idle                 | running  Q |  15 |  8192 ( 2280) | 0x806f180 
	  2 | main                 | bl reply _ |   7 | 16384 ( 2436) | 0x806b180 
	  3 | uart0                | bl rx    _ |   6 |  8192 (  920) | 0x8084ea0 
	  4 | pktdump              | bl rx    _ |   6 | 16384 (  936) | 0x807e3c0 
	  5 | ipv6                 | bl rx    _ |   4 |  8192 ( 1656) | 0x807c1c0 
	  6 | udp                  | bl rx    _ |   5 |  8192 (  936) | 0x80823e0 
	  7 | tapnet               | bl rx    _ |   4 |  8192 ( 1592) | 0x807a1a0 
	    | SUM                  |            |     | 73728 (10756)
```

authmillenon, 2015-07-13:
(tapnet(7) uses the `_nomac_thread`)

OlegHahm, 2015-07-14:
IdleThread seems to be pretty full again.

kaspar030, 2015-07-14:
> ping6 -f -I tap0 <ipv6 address of RIOT node>

I'm using two times 'ping6 -f -I tap0 <ll-address of RIOT node>%tap0>.
Are you using the link layer address, too?

authmillenon, 2015-07-14:
> I'm using two times 'ping6 -f <ll-address of RIOT node%tap0>.
Are you using the link layer address, too?

Nope, and yes I was using two ping6 processes, too.

authmillenon, 2015-07-14:
Oh btw:

> Alright. To rule out nativenet, we need a l2 test application.
Could someone write a shell command that takes <if-pid> <target-mac-addr> <num> [<delay>] [<extra-data>] as arguments,
and sends <num> l2-packets with ethertype 0xffff to <target-mac-addr> via <if-pid> with <delay> us delay between packets.
The payload of the packet should contain the nr of the packet, starting with 0 for the first. If <extra-data> is given, the packet should be filled with that number of garbage bytes (e.g., zeroes).
> 
> Volunteers?

See #3389.

kaspar030, 2015-07-14:
> Are you using the link layer address, too?
> 
> Nope

Different problem then, should open another issue. Could you try two pin6 -f to the link layer address?

authmillenon, 2015-07-14:
As with `ping6 -f %tap0` I get

```
[martine@beutlin RIOT]<3 sudo ping6 -f 3e:e4:ff:2e:99:6c%tap0
unknown host
```

authmillenon, 2015-07-14:
> Different problem then, should open another issue.

Can you point out to me, why you think that this is a different issue?

OlegHahm, 2015-07-14:
> Different problem then, should open another issue. Could you try two pin6 -f to the link layer address?

What does ping6 do if you specify a link layer address?

kaspar030, 2015-07-14:
@authmillenon use the link-layer IPV6 addres suffixed by the interface, not the mac address.

kaspar030, 2015-07-14:
Sorry, I mean link-local...

kaspar030, 2015-07-14:
> Can you point out to me, why you think that this is a different issue?

Well, I'd assume it is a different issue because Oleg and I can
reproducably crash at the same line, and you're seeing a different crash.

authmillenon, 2015-07-14:
With 2x `sudo ping6 -f fe80::3ce4:ffff:fe2e:996c%tap0` I get the SIGTRAP in `__isr_stack()` again. Can't say for sure, that the other bugs won't occure, too. I'm still not convinced, that this is a different issue, because we saw different behaviors on different machines caused by the same bug with native before. What toolchains are you two using? 

```
[martine@beutlin RIOT]<3 gcc --version
gcc (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

kaspar030, 2015-07-14:
I'm on
```
[kaspar@booze riot.7 (msba2-next)]$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-unknown-linux-gnu/5.1.0/lto-wrapper
Target: x86_64-unknown-linux-gnu
Configured with: /build/gcc-multilib/src/gcc-5-20150623/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release --with-default-libstdcxx-abi=c++98
Thread model: posix
gcc version 5.1.0 (GCC) 
[kaspar@booze riot.7 (msba2-working)]$ 
```

authmillenon, 2015-07-14:
(with flags):

```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.4-2ubuntu1~14.04' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04)
```

kaspar030, 2015-07-14:
```
[kaspar@booze riot.9 (master)]$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-unknown-linux-gnu/5.1.0/lto-wrapper
Target: x86_64-unknown-linux-gnu
Configured with: /build/gcc-multilib/src/gcc-5-20150623/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release --with-default-libstdcxx-abi=c++98
Thread model: posix
gcc version 5.1.0 (GCC) 
[kaspar@booze riot.9 (master)]$ 
```

cgundogan, 2015-09-10:
I can't reproduce this. Riot does not crash for me on current master. Fixed?

OlegHahm, 2015-09-10:
Can still reproduce.

kaspar030, 2015-09-15:
May be related to #3341.

OlegHahm, 2015-09-15:
Self-reference?

kaspar030, 2015-09-15:
Ups. #3824.

OlegHahm, 2015-09-15:
Could be, but from the backtrace I would guess that is a different thing.

miri64, 2016-10-13:
Since nativenet does not exist anymore I allowed myself to rename this ;-).

OlegHahm, 2016-11-10:
The last time I tested after #2071 being merged, I was not able to reproduce. Can anyone verify?

miri64, 2016-11-10:
I ran this on both a bridge and a vanilla tap for about 2 min and wasn't able to reproduce.

kaspar030, 2016-11-10:
Me neither. Awesome! Closing...

closed 2016-11-10 by kaspar030