RIOT-OS/RIOT #3346: drone-scripts/build_and_test.sh: individual builds may fail randomly

phiros, 2015-07-09:
I've noticed several times now that individual builds can fail randomly (for example ```example/default``` seems to fail a lot for ```avsextrem```). It looks like something in either ```Make.buildtest``` / ```Makefile.include``` or ```compile_test.py``` prevents examples/tests from being compiled properly when multiple buildtests (for different ```BUILDTEST_MCU_GROUP```s) run concurrently. 
So far I've noticed one case in which the linker complained that an archive was missing.
Maybe ```make``` is executing something in parallel which it shouldn't. 
The fix for  #3304, which I am working on, will properly reveal the root of this issue.

This also affects our alternative CI system. Currently, it uses this script to build / test RIOT.

miri64, 2016-10-17:
Did anyone encounter this issue still? @kaspar030 @OlegHahm @gebart @haukepetersen @cgundogan?

OlegHahm, 2017-01-15:
Worksforme.

closed 2017-01-15 by OlegHahm