RIOT-OS/RIOT #3350: Strider: race?

gebart, 2015-07-09:
I got the following on my strider run at https://ci.riot-os.org/riot-os/riot/job/559ede2d13d80bee08e15dd9

```
fatal: Unable to create '/home/strider/workspace/.git/index.lock': File exists.

If no other git process is currently running, this probably means a
git process crashed in this repository earlier. Make sure no other git
process is running and remove the file manually to continue.
Failed to merge in the changes.
```

Is there some race between multiple concurrent builds? Does every parallell build have its own workspace? It seems strange that a lock file would already exist if it was a pristine clone.

gebart, 2015-07-09:
I found this which may be related: https://github.com/Strider-CD/strider/issues/745, https://github.com/Strider-CD/strider/pull/832

phiros, 2015-07-09:
I just checked the logs there is no mention of an error (so I don't think that Strider-CD/strider#745, Strider-CD/strider#832 are related to this). However, I did notice that each ```BUILDTEST_MCU_GROUP``` operates on the same directory.
#3352 should fix this problem by giving each ```BUILDTEST_MCU_GROUP``` a separate workspace.

phiros, 2015-07-11:
Fixed in #3352

closed 2015-07-11 by phiros