RIOT-OS/RIOT #3351: Strider: running test has caption "default"

gebart, 2015-07-09:
[image]

The text just says "default" for the strider build.

phiros, 2015-07-10:
Seems like the github-status plugin does not set the context parameter.
I will change this soon.

BTW: Do you have any suggestions for the text? The best I came up with
is "RIOT-OS/Strider-CD".

gebart, 2015-07-10:
Is it possible/desired to have other types of checks as separate strider
runs?
Strider-CD/compile-test?

Maybe we can use separate runs as a way to get a view of the build matrix
directly inside the PR threads on github, or is that more difficult to
maintain?
Strider-CD/static checks, Strider-CD/compile-test/cortex-m4 etc.

OlegHahm, 2015-07-10:
If this is possible, I think it would be very helpful.

phiros, 2015-07-10:
With the current version of strider it is not possible to have multiple projects per github repo.
However, I started to work on a build matrix plugin anyways. Maybe I can integrate your idea for
several status messages.
We just have to keep in mind that there is a rate limit for the Github API end-point.

kaspar030, 2015-07-10:
> We just have to keep in mind that there is a rate limit for the Github API end-point. 

Github might raise our limit, they say they do if you can explain why
you need it.

phiros, 2015-07-10:
Cool. That is good to know

OlegHahm, 2015-12-04:
Strider is gone - for now.

closed 2015-12-04 by OlegHahm