RIOT-OS/RIOT #3355: Provide getopt for platforms that don't support it

authmillenon, 2015-07-10:
newlib provides `getopt.h` which would implementing commands like `ping6` far easier. Sadly msp430-libc and avr-libc do not provide this header. Conclusively, a lightweight implementation for those platforms would be much appreciated.

N8Fear, 2015-07-11:
Did you consider to take the getopt stuff from musl? I think musl is a very neatly written libc implementation. It's also relatively modular and doesn't assume too much. It's also MIT-licensed so license shouldn't be a problem either.

MohmadAyman, 2015-10-16:
I've wrote the lib for the avr, I used some of musl's code and some public domain code plus my own code, so about the licence thing, is there a problem with that -using MIT and public domain code-, do I mention who's code did I use, even thought I did modify like half of it ?
Forgive my so many questions =D, I'm new to riot and to open source development in general.
Thanks.

authmillenon, 2015-10-16:
About MIT: there are several variants of the MIT license (most notably the Expat and the X11 license, both are compatible to (LGPLv2.1)[http://www.gnu.org/licenses/license-list.html]). If it is compatible to LGPLv2.1 it should be no problem to include them. Public Domain should be no problem either.

MohmadAyman, 2015-10-18:
thanks for the info about the licensing thing.
I've started a pull request #4106 . but i'm still having some doxgyen errors right now, i'll try to fix them asap.

sudarshansrinivasan, 2017-04-01:
Hi , 
I am complete newbie to RIOT & open source . Currently  looking at the PRs labeled "Newbie-Task-Candidate".
I wanted to know if this PR# is still open  to work on  and suggestions how proceed with it?
I see some changes were submitted for review in https://github.com/RIOT-OS/RIOT/pull/4106.

Thanks

miri64, 2017-04-02:
@sudarshansrinivasan for avr-libc have a look at the [comment by @N8Fear above](https://github.com/RIOT-OS/RIOT/issues/3355#issuecomment-120600215). The implementation (the c-file) should go to `cpu/atmega_common` and the header to `cpu/atmega_common/include`. The idea is to implement a feature interchangable with POSIX's getopt functionality: https://docs.python.org/3/library/getopt.html.

@gebart how is the status for this on msp430 (+newlib?)?

miri64, 2017-04-02:
(`getopt` suffices for now, no need for `getopt_long`)

gebart, 2017-04-02:
I don't know about msp430 libc, but I guess newlib on msp430 has it if newlib on arm has it.

miri64, 2017-04-02:
I have to say since #6778  I'm pretty confused how newlib and msp430 work now ;-). But that's OT, I know.

virajsahai, 2018-05-22:
Hello..
I have written an implementation for this. Can anyone tell me how to test it?

miri64, 2019-08-10:
Fixed by https://github.com/RIOT-OS/RIOT/pull/9538

miri64, 2020-02-11:
Fixed by partly by #12457?

miri64, 2020-08-05:
`avr-libc` still does not support this AFAIK.

maribu, 2022-04-19:
Is only avrlibc missing getopt? Now that avrlibc is actively maintained again, one could try to upstream getopt instead.

Anyway, I think getopt is part of the POSIX environment that we can provide to ease life for developers new to RIOT. So I guess we should reopen this.