RIOT-OS/RIOT #3363: Proposal for directory structure for automated tests

A-Paul, 2015-07-10:
Define a directory under `$RIOTBASE/tests/` and outside `tests/unittests` where sources for automated tests can be placed. It has turned out in discussions [#3296](https://github.com/RIOT-OS/RIOT/pull/3296#issuecomment-118338031), [#3307](https://github.com/RIOT-OS/RIOT/pull/3307#discussion_r33860255) that the established *unittests* becomes insufficent for extended testing.

Motivation
----------

The agreement of a seperate directory structure should face the following difficulties.

### tests/unittests ###

The tests in this directory are build together in one blob. This is implicating some weaknesses. 

+ With the increasing number of test cases the memory ressources of some boards are exhausted.
+ It prevents the desired isolation of the running test cases.
+ Necessary control of preconditions can not be archieved for test sequences that would require reinitializing (e.g `ng_ipv6_init()`)

### tests ###

The directory is, beside of *unittests* home of several more or less freely named subdirectories for different test scenarios. It is not adviseable to mix the automated tests inbetween.

+ Increases naming collisions or encourages creative naming to prevent them.
+ Makes it difficult to keep track of locations.

Proposal
--------

The idea is to name name subdirectory of tests(lets say *'automated'*) and put the tests in subdirectories. These may reflect the location of either the headers or the units source files to follow the "principle of least suprise".
Every *unit* or *test object* has further subdirs for sepeated test which require reflashing, etc...

### Example for resulting structure ###
(Reflecting loctions of C files)

    tests/automated/core/atomic/
    tests/automated/core/...
    tests/automated/core/thread/
    tests/automated/...
    tests/automated/sys/auto_init/auto_init/
    tests/automated/sys/...
    tests/automated/sys/net/application_layer/ng_zep/ng_zep/
    tests/automated/sys/net/...
    tests/automated/sys/net/transport_layer/ng_udp/ng_udp/

authmillenon, 2015-07-11:
Though I'm much in favor of structuring the automated tests better (though we should IMHO even more subdivided by the means of testing [unittests with embUnit, tests via shell using pexpect or similar tools, etc.]), we also should consider that different combinations of modules can lead to slightly different behaviors (that was the original idea behind putting all unittests for all modules in one application). So just dividing them by module does not do the trick.

PS: the CI-Infastructure label is for issues related to the CI infrastructure (the CI server and its software etc.) not for the tests, so I removed the label.

OlegHahm, 2015-07-13:
I like the proposal, too.

@authmillenon, did I get it right that you are also in favor of having this proposal implemented?

authmillenon, 2015-07-13:
I'm in favor of better structuring (and sub-dividing) of the automated tests. But I'm not sure if the current proposal makes sense with regards to more complicated tests than just module-wise tests (e.g. how does IPv6 behave with and without 6LoWPAN etc.).

OlegHahm, 2015-07-13:
I'm not sure, I get this point. If tests are covering only one module or more than one should be unaffected by the question of having a subfolder for automated tests or do I get you wrong?

authmillenon, 2015-07-13:
What I say is: What if the test is specifically for 2 or more modules and only makes sense if those modules are involved?

OlegHahm, 2015-07-13:
Sorry, I still don't understand the problem.

OlegHahm, 2015-07-13:
Ah, after a offline-explanation I got it. Yes, I agree: putting automated tests into subfolders is a good idea, "cloning" RIOT's folder structure in this tests folder seems to be a) overkill and b) not cover all test cases.

A-Paul, 2015-07-14:
> Ah, after a offline-explanation I got it. Yes, I agree: putting automated tests into subfolders is a good idea, "cloning" RIOT's folder structure in this tests folder seems to be a) overkill and b) not cover all test cases.

@OlegHahm 
Regarding *"a)"*: Is it the depth or the width of the directory structure?

Regarding *"b)"*:

> I'm in favor of better structuring (and sub-dividing) of the automated tests. But I'm not sure if the current proposal makes sense with regards to more complicated tests than just module-wise tests (e.g. how does IPv6 behave with and without 6LoWPAN etc.).

@authmillenon
In terms of testing you are correctly defining the next test level after the unittests successfully passed, the integration test. And it's a sensible question wheather to mix unit- and integration tests in one directory tree.

***Footnote:*** Integration testing also should have a somewhat defined testplan as *shurely* the number of integration tests shouldn't be the power set of all seperate units.

authmillenon, 2015-08-03:
#2231 might be a good starting point.

miri64, 2016-10-17:
Any progress here?

OlegHahm, 2017-01-15:
Stalled?

closed 2019-09-10 by stale[bot]