RIOT-OS/RIOT #3365: ng_pktbuf_add: possible packet corruption in case of internal error

daniel-k, 2015-07-10:
[Link to source](https://github.com/RIOT-OS/RIOT/blob/d2d4e69d8acdd4988d06917b7d89e1aa0e1b97cd/sys/net/crosslayer/ng_pktbuf/ng_pktbuf.c#L275)

```c
pkt->size -= size;
DEBUG("pktbuf: resize pkt->size to %u\n", (unsigned)pkt->size);
pkt->data = (void *)(((uint8_t *)pkt->data) + size);
```
...
```c
if (!_pktbuf_internal_add_pkt(pkt->data)) {
    DEBUG("failed (freeing %p)\n", (void *)new_pktsnip);
    _pktbuf_internal_free(new_pktsnip);

    return NULL;
}
```

In case this fails it leave `pkt` altered. 

The only thing is that I don't really understand the circumstances for ` _pktbuf_internal_add_pkt(pkt->data) == false`. I get this when I supply `size` so that `size == pkt->size`, i.e. trying to dissect a pktbuf in one part that has all the data and one part that has `size = 0`.

authmillenon, 2015-07-11:
`size == 0` will become illegal anyways when we will redo the `ng_pktbuf` API some day. Or do you have a valid point why this is useful? Even though you described your use case I'm not sure *why* you do it.

`_pktbuf_internal_add_pkt()` returns `false` when `pkt->data` is not part of a packet in the packet buffer, so I'm not sure how this is related to the size parameter (maybe `pkt->data` becomes some illegal value). 

Fix in #3367

daniel-k, 2015-07-11:
> Or do you have a valid point why this is useful? Even though you described your use case I'm not sure why you do it.

I have a header that specifies the type of my payload. Some types have payload, some have not. When I "add" the header while there's no payload behind, I `pkt_buf_add` something on existing data with the size of the existing data.
If it's meant to be used otherwise please tell, but that's what I figured out it's supposed to do.

daniel-k, 2015-07-11:
> `_pktbuf_internal_add_pkt()` returns false when pkt->data is not part of a packet in the packet buffer

At the beginning you check
```c
if (size > pkt->size) {
```
So this would never happen, would it?

authmillenon, 2015-07-11:
If there is no payload, why do you add a `ng_pktsnip_t` for that?

authmillenon, 2015-07-11:
> At the beginning you check
> 
>     if (size > pkt->size) {
> So this would never happen, would it?

I don't get what this has to do with `pkt` being in the packet buffer...

daniel-k, 2015-07-11:
> I don't get what this has to do with pkt being in the packet buffer...

Well if `size > pkt->size` you cancel anyway, so `0 <= size <= pkt->size`. 

As `pkt->data` is already in the packet buffer, incrementing it by something smaller than `pkt->size` will surely be in the packet buffer, too. 

So there's just the case when `size == pkt->size` where `pkt->data` is not inside the packet buffer anymore. And this is what happens to me.

authmillenon, 2015-07-11:
> As pkt->data is already in the packet buffer, incrementing it by something smaller than pkt->size will surely be in the packet buffer, too.

Why would it be? E. g.

```C
ng_pktsnip_t pkt;
pkt.data = "12345";
pkt.size = sizeof("12345");
```

is not in the packet buffer, but one can still call `ng_pktbuf_add()` with it (not sure if it makes sense, but the check in `_pktbuf_internal_add_pkt()` ensures it does not do weird things to it).

daniel-k, 2015-07-11:
> If there is no payload, why do you add a ng_pktsnip_t for that?

I'm dissecting packets comming out of netdev. The pktsnip I get has `pkt->type` UNDEF and `pkt->next->type` NETIF.

Now I want to analyse the UNDEF payload. I expect 2 bytes that are the header of my MAC layer. After these 2 bytes of header there might be any data or not. At that moment, I don't know this yet. Depending on what the header states I'll know then if there is more data to expect.

If you want to see the code, [here](https://github.com/daniel-k/RIOT/blob/ef3cef5f2fd4224dd0fc83edde4e7dc3cb1a85c9/sys/net/link_layer/ng_lwmac/rx_state_machine.c#L103)

daniel-k, 2015-07-11:
> is not in the packet buffer, but one can still call ng_pktbuf_add() with it (not sure if it makes sense, but the check in _pktbuf_internal_add_pkt() ensures it does not do weird things to it).

Okay. I didn't think of the possiblity that `pkt->data` might point to somewhere not in the packet buffer. This way your fix #3367 makes sense now. I always thought that this was supposed to be some kind of bounds check after moving `pkt->data`.

authmillenon, 2015-07-11:
Aaaaahh… somehow I was assuming you added the payload with `ng_pktbuf_add()`, not the header. Now I get it... I'm honestly not sure how the packet buffer handles this with #3367. Can you check?

daniel-k, 2015-07-11:
> Now I get it... I'm honestly not sure how the packet buffer handles this with #3367. Can you check?

Will do on Monday. Didn't take my hardware at home :)

PR #3367 (2015-07-13): ng_pktbuf: fix possible packet corruption in case of internal error
Fixes #3365
commit d1b34aec (2015-07-13): ng_pktbuf: fix possible packet corruption in case of internal error

Fixes #3365
closed 2015-07-13 by authmillenon