RIOT-OS/RIOT #3366: periph/i2c: handle NACK

LudwigOrtmann, 2015-07-11:
The I2C peripheral driver interface does not currently handle a slave `NACK`.

As there are I2C devices which have `NACK` as a valid part of the communication process, the current interface (which only returns `-1` on `undefined device given`) can not cope with these devices.

I'd suggest adding a `-2` for `no ACK received` to the interface documentation and changing the driver implementations accordingly.

haukepetersen, 2016-02-29:
should be fixed with #4926

LudwigKnuepfer, 2016-03-01:
As far as I see #4926 only changes the interface, the implementations still need to be changed to actually handle a NACK.

haukepetersen, 2016-03-01:
of course, but if there is an agreement on #4926, the implementations will follow

lebrush, 2016-03-02:
Related #4736 (there it's already returning -2)

kYc0o, 2018-07-17:
Is the I2C rework handling this? As far as I can tell, yes, so it would be good if it's referenced somewhere.

miri64, 2018-07-17:
@MrKevinWeiss @aabadie @dylad?

dylad, 2018-07-17:
In the I2C refactoring, when a NACK is received from the device, the I2C function will return -EIO so it is handled as an error.
If a NACK can be use as part of the protocol (not handled as error) then we should add some more work. We must first identify which sensors are using NACK this way.

MrKevinWeiss, 2018-07-17:
We should distinguish between a DATA NACK and an ADDR NACK.
The HDC1010 temperature/humidity sensor will NACK the address if trying to read data that isn't read and will NACK the data byte if trying to write to a readonly register such as a serial number.

According to the documentation 
```
* @return                  -EIO When slave device doesn't ACK the byte
* @return                  -ENXIO When no devices respond on the address sent on the bus
```
So it appears like both cases are covered and the user can handle the logic in the application.  Maybe not all cpu's currently support the error codes (some may just timeout) but everything is available to do so.

The [SI7006](https://www.silabs.com/documents/public/data-sheets/Si7006-A20.pdf) allows for No Hold Master Mode which after sending a measure command the idea is to poll the slave address, getting nacks, until an ack is given.  On page 20 of the datasheet it gives an example.  From the looks of this it may be a problem since it seems to require a repeated start and (unless the I2C_NOSTOP flag is set) the current implementation would be to send a stop if a NACK is received.  However this sequence could  cause some lockup problems in other sensors.

MrKevinWeiss, 2018-08-02:
Now that the rework is here should we do something with this PR?

MrKevinWeiss, 2018-08-07:
@dylad Do you want to investigate further? I tend to think special i2c protocols should be handled in the higher levels.  We should only support the read/write reg functions since they are common.  If we all agree then I think we can close this issue.

dylad, 2018-08-07:
@MrKevinWeiss We can investigate it further but I'm currently in holidays so you'll have to start without me.

MrKevinWeiss, 2019-08-10:
I have a think I fixed this.  The HiL tests should be checking error codes.  I think we can close it @dylad do you agree?

MrKevinWeiss, 2019-11-22:
Yup, seems like a forgotten issue!

closed 2019-11-22 by MrKevinWeiss