RIOT-OS/RIOT #3371: unittests_constants: TEST_STRING8 is 7 chars long

gebart, 2015-07-13:
https://github.com/RIOT-OS/RIOT/blob/master/tests/unittests/common/unittests-constants.h#L22
is only 7 characters long if not counting the terminating NULL byte because of the double backslashes only resulting in a single backslash in the string.
From looking at the other constants it seems like the string is expected to be 8 chars excluding the terminating NULL.

Also worth mentioning is that the 64 char string is 68 chars + terminating NULL.

gebart, 2015-07-13:
feel free to close if this is intentional.

authmillenon, 2015-07-13:
Nope that was definitely not intentionally and explains why ASAN was complaining about in one of the lines I "fixed" in #3369 by not using `TEST_STRING8` ;-). Will do a PR.

PR #3374 (2015-07-13): unittests: fix length of test data strings
Fixes #3371 

Strings have new values since I used a PRNG to generate them (as I did originally).

```python
#!/usr/bin/env python
import random
print(''.join(chr(random.randint(32,127)) for _ in range(8)))
print(''.join(chr(random.randint(32,127)) for _ in range(64)))
```
commit 888bdd70 (2015-07-13): unittests: fix length of test data strings

Fixes #3371
closed 2015-07-13 by PeterKietzmann