RIOT-OS/RIOT #3377: boards/qemu-arm: Add support for lm3s6965evb board for testing in qemu-system-arm

gebart, 2015-07-13:
lm3s6965evb and lm3s811evb are both Cortex-M3 based and supported as emulated machines in QEMU. Adding support for these boards will let us run unittests etc on virtual Cortex-M3 hardware inside Travis/Strider to catch any ARM Cortex specific issues. e.g. tests on the scheduler code, syscalls, memory allocation, anything not tied to a particular vendor..

phiros, 2015-07-23:
Porting this would definitely be beneficial. However, I've a lot to do right now
with the CI system. Might take a few weeks until I can port this (I hope I can do it earlier)

BTW: this would also offer us the chance to test network related stuff on a emulated platform
which is much closer to a "real" board than native.

latsku, 2015-09-17:
I will just leave this link here. http://gnuarmeclipse.github.io/qemu/ 
I haven't tested that at all, but the projects seems to fit more than perfectly on this one.

kousu, 2018-10-27:
@gebart thank you! That is very helpful! I realize this is out of your purview a bit, thanks for taking the time to help me out :)

gebart, 2018-10-27:
@kousu no problem! Tell us if you have any progress in emulating RIOT on cortex m

nghiaducnt, 2019-04-08:
@gebart I have just learned RIOT OS with QEMU's lm3s6964evb recently. Basically, the RIOT OS works well with QEMU. However, to work with advanced features(like timer or networking) QEMU's Stellaris needs some fixes.  Here is my work https://nghiaducvn.wordpress.com/. When GPTimer and Ethernet device is ready, it is really awesome to see and experiencing networking applications over TAP devices.

jcarrano, 2019-04-09:
@nghiaducnt I just read your blog post. Pretty awesome! Having qemu support with networking would improve our testing process a lot.

I would suggest that you do not wait to have the GPTimer and Ethernet working to make a PR. Instead, PR the basic board support as soon as you think it is good enough, and keep on working on the more advance functionality while we review it. This way you will have something to show and hopefully spark interest amongst the RIOT community.

nghiaducnt, 2019-04-10:
@jcarrano my original intention was just to have basic features. However, it grew more and more interesting when it came as a whole system, just could not stop to solve the problem. With QEMU, it is easy for me to penetrate the system. Just like GPTimer's device driver or injecting unusual network traffic to verify network stack(then security), sometimes I saw the system unresponsive but I couldn't dive deep to find the reason(QEMU was not in debug mode at that time). My next experiment, I would add/fix the simulation of clock/random number generator/security IP block in QEMU. I would like to see RIOT can support well network application with security features. I will keep you guys posted.

closed 2019-11-12 by stale[bot]