RIOT-OS/RIOT #3378: Travis: Use container infrastructure

gebart, 2015-07-13:
> This job ran on our legacy infrastructure. Please read our docs on how to upgrade

http://docs.travis-ci.com/user/migrating-from-legacy/?utm_source=legacy-notice&utm_medium=banner&utm_campaign=legacy-upgrade

Using this would require pre-packaging a lot of binaries because we can not use `sudo` any more.

A related question is: Will Strider become our primary CI system in the near future?

phiros, 2015-07-14:
Strider will only be adopted as the main CI system if the community decides it makes sense
to do so. For the time being we will continue to use Travis in parallel with strider.
Therefore, I think it makes sense to also invest some time into the travis scripts.
Meaning: I will take a look at the new container infrastructure.

miri64, 2016-10-17:
We don't use travis anymore, so let's close this

closed 2016-10-17 by miri64