RIOT-OS/RIOT #3383: CI: Publish buildsizes results on a web page

gebart, 2015-07-13:
Continuation of the discussion at the end of #3368:

@OlegHahm: 
> Would be cool to generate numbers automatically for the web page.

@gebart:
>@OlegHahm :+1: I would like to expand on that thought and generate build sizes for each platform for each of the examples (hello-world, ng_networking, rpl_udp, default) for each master commit in order to see how it evolves over time. Takes a lot of resources (and therefore time) to compile everything though.

@OlegHahm:
>Since the CI is doing this more or less anyways, wouldn't it make sense to generate these numbers on the CI system(s) and upload them somewhere? @phiros, what do you say?

gebart, 2015-07-13:
I was thinking of pushing this kind of information to a thingspeak.com channel and presenting the numbers on a web page.

OlegHahm, 2015-07-13:
I have no experience with thinkspeak.com, but I'll take a look. @backenklee, maybe you're interested in this topic?

phiros, 2015-07-14:
I am not super sure about thingspeak (AFAIK each channel only allows for 8 different fields). However,
it would be nice to have some kind of charting website on which we can compare current master against pull requests (grouped by BUILDTEST_MCU_GROUP). I will look into this tomorrow.

aabadie, 2017-07-07:
This can be found in Murdock results now. Closing this one.

closed 2017-07-07 by aabadie

kaspar030, 2017-07-07:
> This can be found in Murdock results now. Closing this one.

Well, murdock collects the data, but there's no visualization for it...

kaspar030, 2017-07-07:
> I was thinking of pushing this kind of information to a thingspeak.com channel and presenting the numbers on a web page.

I just checked their terms, without paying they allow 8000msgs/day. Sounds like a lot, but we currently have ~13k builds, each with text, data, bss values. This number will grow, and we'll soon add both power measurements and benchmark results...

gebart, 2017-07-07:
They might offer a discount for RIOT for being open source and IoT, did anyone ask them?