RIOT-OS/RIOT #3391: doc: wrong hierarchy level for some modules

OlegHahm, 2015-07-14:
For some weird reasons some modules (e.g. 6LoWPAN or IEEE802.15.4e) show up at the wrong hierarchy level at http://doc.riot-os.org/modules.html. Building the documentation locally seems to generate the expected hierarchy. The documentation on the web page is generated by doxygen version 1.8.8 - locally I have 1.8.9.1. Can someone cross check?

authmillenon, 2015-07-14:
I've got 1.8.6 and the same result as on the website.

gebart, 2015-07-14:
I will check this later.
Is it possible to upgrade the website Doxygen version if there is a bug?

authmillenon, 2015-07-14:
`net_if` (Network interfaces) is also at the wrong spot on the website. BTW: both 6LoWPAN and IEEE802.15.4 are the old network stack's versions of those modules. The new ones are in the right hierarchy.

gebart, 2015-07-14:
@authmillenon I'm guessing without looking at the code that there is a problem with the `@ingroup` commands or similar.

authmillenon, 2015-07-14:
Nope, all have `@ingroup net` in their respective `.h` files.

OlegHahm, 2015-07-14:
Ok, then it seems to be indeed a doxygen bug. Will try to upgrade on the server.

OlegHahm, 2015-07-14:
Unfortunately there doesn't seem to be a newer version of doxygen in backports.

OlegHahm, 2015-07-15:
Ok, the original issue has been fixed by updating doxygen, but still "Atmel SAM3X8E Definitions", "RPL source routing header extension", "SAM3XA Components", and "dev_eth auto setup" are showing up on the top level.

OlegHahm, 2015-07-15:
Fixed in #3412.

authmillenon, 2015-07-15:
(slightly related: would be great if the soon to be phased out modules would be marked as such in their description)

OlegHahm, 2015-07-15:
I plan to do update the documentation on Friday in the train to Prague.

PR #3412 (2015-07-21): doxygen: documentation hierarchy fixes
fixes #3391
closed 2015-07-21 by OlegHahm