RIOT-OS/RIOT #3392: Strider: Automatic testing of native on different platforms

gebart, 2015-07-14:
I would like to see automatic testing of the native platform on other systems than Debian+Linux+x86(or x86_64).

Suggestions for tests:
 - FreeBSD
 - OSX
 - armv7a-hardfloat-linux-gnueabihf (Cortex-A* linux)
 - AArch64 (armv8) (64 bit ARM)

authmillenon, 2015-07-14:
This is planned anyways.

> * armv7a-hardfloat-linux-gnueabihf (Cortex-A* linux)

AFAIK it is possible to build native on Raspberry Pi, so this should be easy.

> * AArch64 (armv8) (64 bit ARM)

We don't even have (and probably never will) support for x86_64 Linux (I'm not even sure if that would make sense currently, since all real hardware platforms we target only go up to 32bit), so why add support for this platform.

gebart, 2015-07-14:
@authmillenon fair enough

miri64, 2016-10-17:
Renamed

miri64, 2016-10-17:
Also some relation to #5201.

OlegHahm, 2017-01-15:
@cgundogan, @smlng, @kaspar030, this is currently happening in the CI TF, right?

smlng, 2017-01-15:
Yes, macOS is WIP for Jenkins see #6341, and I just created #6371 to activate native compile on Raspberry Pi ...

smlng, 2019-09-09:
MacOS is not supported as RIOT native does not run on 64Bit CPUs (yet)

miri64, 2019-09-09:
We still could consider testing on FreeBSD, as `native` was originally developed with support for it.

kaspar030, 2019-09-09:
> We still could consider testing on FreeBSD, as `native` was originally developed with support for it.

And let's not forget armhf. There are plenty of RasPi's that could test something.

miri64, 2019-09-09:
@kaspar030's argument is to me more to the point than my (IMHO very weak) argument, so I think we should re-open.

smlng, 2019-09-09:
hmm, every time I brought this up (test on other platforms and compilers) someone told me that riot docker is the reference and we cannot test on more - at least for now. Also there is and was no movement in changing this for a long time.

miri64, 2019-09-09:
> hmm, every time I brought this up (test on other platforms and compilers) someone told me that riot docker is the reference and we cannot test on more - at least for now. Also there is and was no movement in changing this for a long time.

I at least disagree with that sentiment (but don't have the time or resources atm to put it in place myself). However, see https://github.com/RIOT-OS/riotdocker/issues/66

smlng, 2019-09-09:
okay then we need to clarify this a bit, what would be the goal? Something like, nightlies on more platforms? I think that is possible but for every PR might be a bit out-of-scope and also above the available resources. Also MacOS is out for now.

Which platforms do we primarily want (bc ideally: all of them):

- BSD
- ARM (like Raspi, which one?)
- other Linux-Distros: CentOS, Fedora, Debian, Arch?

And then: which compiler, the one from the official repos or newest from third-party repos or even latest gcc/lvm build from scratch?

miri64, 2019-09-09:
> okay then we need to clarify this a bit, what would be the goal? Something like, nightlies on more platforms? I think that is possible but for every PR might be a bit out-of-scope and also above the available resources. Also MacOS is out for now.

Agreed on both points.

> Which platforms do we primarily want (bc ideally: all of them):
> 
> * BSD
> * ARM (like Raspi, which one?)
> * other Linux-Distros: CentOS, Fedora, Debian, Arch?

as you said: Ideally all of them, however I think Raspi should have priority, then some non-apt Linux distro (given that many of us use Arch I'd say that and maybe Fedora). BSD, though it would be great to have, should have the least priority, as the current state is unclear (I think it was last tested when native was implemented).

> And then: which compiler, the one from the official repos or newest from third-party repos or even latest gcc/lvm build from scratch?

Official compiler should have first priority.

maribu, 2022-09-16:
@kaspar030 Does the pifleet also test `native` (that is, on Linux/ARM)?

kaspar030, 2022-09-16:
> @kaspar030 Does the pifleet also test `native` (that is, on Linux/ARM)?

Nope, not at the moment. They're also not really equipped to compile much themselves.