RIOT-OS/RIOT #3408: tests/libfixmath_unittests: broken

phiros, 2015-07-14:
It seems like ```tests/libfixmath_unittests``` is broken right now.

```
/tests/libfixmath_unittests$ make distclean 
rm -rf /home/philipp/RIOT/tests/libfixmath_unittests/bin/native/libfixmath-src/ /home/philipp/RIOT/tests/libfixmath_unittests/bin/native/libfixmath-headers/
rm -rf /home/philipp/RIOT/pkg/libfixmath/checkout/trunk-r91
```

```
/tests/libfixmath_unittests$ make buildtest BUILDTEST_VERBOSE=1 BUILD_IN_DOCKER=1 BUILDTEST_MCU_GROUP=arm7
Launching build container using image "riot/riotbuild:latest".
docker run --rm -i -t -u "$(id -u)" \
	    -v '/home/philipp/RIOT:/data/riotbuild/riotbase' \
	    -v '/home/philipp/RIOT/cpu:/data/riotbuild/riotcpu' \
	    -v '/home/philipp/RIOT/boards:/data/riotbuild/riotboard' \
	    -v '/home/philipp/RIOT:/data/riotbuild/riotproject' \
	    -e 'RIOTBASE=/data/riotbuild/riotbase' \
	    -e 'RIOTCPU=/data/riotbuild/riotcpu' \
	    -e 'RIOTBOARD=/data/riotbuild/riotboard' \
	    -e 'RIOTPROJECT=/data/riotbuild/riotproject' \
	    -e 'BUILDTEST_MCU_GROUP=arm7' -e 'BUILDTEST_VERBOSE=1' \
	    -w '/data/riotbuild/riotproject/tests/libfixmath_unittests/' \
	    'riot/riotbuild:latest' make buildtest
/data/riotbuild/riotbase/Makefile.include:232: target '/data/riotbuild/riotproject/tests/libfixmath_unittests/bin/native/libfixmath.a' given more than once in the same rule
Building for avsextrem .. failed
Building application "libfixmath_unittests" for "avsextrem" with MCU "lpc2387".

svn checkout -r 91 http://libfixmath.googlecode.com/svn/trunk/ /data/riotbuild/riotbase/pkg/libfixmath/checkout/trunk-r91
cp: cannot stat '/data/riotbuild/riotbase/pkg/libfixmath/checkout/trunk-r91/unittests/*.[ch]': No such file or directory
Makefile:35: recipe for target '/data/riotbuild/riotproject/tests/libfixmath_unittests/bin/avsextrem/libfixmath-unittests-src/Makefile' failed
make[1]: *** [/data/riotbuild/riotproject/tests/libfixmath_unittests/bin/avsextrem/libfixmath-unittests-src/Makefile] Error 1
make[1]: *** Waiting for unfinished jobs....
A    checkout/trunk-r91/contrib
A    checkout/trunk-r91/contrib/fix16_fft.c
[...]
Checked out revision 91.
/data/riotbuild/riotbase/Makefile.include:233: recipe for target '/data/riotbuild/riotproject/tests/libfixmath_unittests/bin/avsextrem/libfixmath.a' failed
make: *** [/data/riotbuild/riotproject/tests/libfixmath_unittests/bin/avsextrem/libfixmath.a] Error 2
```

phiros, 2015-07-15:
I should mention that subsequent builds don't fail (as long as you don't clean).
My best guess is that this is a concurrency issue in the Makefile.
Make probably doesn't know that the process "downloading the source of libfixmath"
and compiling "libfixmath_unittest" has to be executed sequentially. This can be achieved
by making one target depend on the other.

gebart, 2015-07-15:
I think I found it.

gebart, 2015-07-15:
Sorry, I should have read your post more thoroughly, I found the same issue that you described..

phiros, 2015-07-16:
I looked into this bug but couldn't identify the exact location of the problem. The only thing I can say for sure is that it is a target dependency problem.

BTW: I took a quick look at libfixmath itself. It uses a MIT license. @OlegHahm and @authmillenon why do we even bother with this error-prone process of downloading the sources via svn and patching them etc. if we can just take the source and make it a regular sys module (like we did with embunit which is also licensed under MIT)?

OlegHahm, 2015-07-16:
Well, if libfixmath isn't actively developed any more, we can surely do so.

Kijewski, 2015-07-16:
:+1: for adopting the code

miri64, 2016-10-17:
I couldn't reproduce. Is this still an issue?

OlegHahm, 2017-01-15:
Worksforme.

closed 2017-01-15 by OlegHahm