RIOT-OS/RIOT #3411: Implement support for 6LoWPAN IPV6 Header Compression

tfar, 2015-07-15:
The following log is produced when sending the attached PCAP (download the image and change the extension from png to pcapng) to RIOT. It was captured during testing with this guide:
https://github.com/RIOT-OS/RIOT/wiki/How-to-install-6LoWPAN-Linux-Kernel-on-Raspberry-Pi

```
2015-07-15 10:58:20,934 - INFO # nomac: sending pkt 0x200019f0 to PID 5
2015-07-15 10:58:20,937 - INFO # 6lo: NG_NETDEV_MSG_TYPE_RCV received
2015-07-15 10:58:20,940 - INFO # 6lo: dispatch 7b ... is not supported
2015-07-15 10:58:20,943 - INFO # 6lo: waiting for incoming message.
```

````
> sha1sum raspberry_pi_linux_6lowpan_header_compression.pcapng 
d0b2bdf99b671a3fa6593abcb9436bcd308a36d8  raspberry_pi_linux_6lowpan_header_compression.pcapng
```
![raspberry_pi_linux_6lowpan_header_compression](https://cloud.githubusercontent.com/assets/254701/8695213/0afff71c-2ae4-11e5-9228-7cbd66132c76.png)

OlegHahm, 2015-07-15:
Probably you should include `USEMODULE += ng_sixlowpan_iphc` in your Makefile.

I think one should rename this issue to "Update the Raspberry Pi 6LoWPAN tutorial" or something like this.

(But be aware that IPHC hasn't been tested a lot so far.)

tfar, 2015-07-15:
@OlegHahm I tested this with the ng_networking example and ```USEMODULE += ng_sixlowpan_iphc``` was added to the Makefile.

authmillenon, 2015-07-15:
Yes, it's as @OlegHahm stated. The reason is is not activated yet is basically because it is in pre-alpha state. However, feel free to contribute big fixes for the bugs you stumble upon :)

OlegHahm, 2015-07-15:
>  I tested this with the ng_networking example and USEMODULE += ng_sixlowpan_iphc was added to the Makefile.

You tested with the module and still got the warning?

tfar, 2015-07-15:
> You tested with the module and still got the warning?

Correct. As you can see at https://github.com/RIOT-OS/RIOT/blob/master/sys/net/network_layer/ng_sixlowpan/ng_sixlowpan.c#L91 only uncompressed headers are currently supported.

OlegHahm, 2015-07-15:
This check seems to be broken indeed.

PR #3414 (2015-07-15): ng_sixlowpan: fix IPHC dispatch check
Fixes #3411
closed 2015-07-15 by authmillenon