RIOT-OS/RIOT #3413: shell: inputs for ng_networking example does not react anymore

BytesGalore, 2015-07-15:
seems that
```
commit a7f3a3d17fd40f43ab4cd645116224cd1763012a
Merge: 30b2bf6 49ea7a3
Author: Oleg Hahm <oleg@hobbykeller.org>
Date:   Tue Jul 14 23:09:07 2015 +0200

    Merge pull request #3161 from kaspar030/newlib_factor_out_uart_stdio
   
    sys: factor stdio uart functions out of newlib's syscalls
```
broke something for the `samr21-xpro` board.
The `ng_networking` example doesn't react on inputs from this commit on.
Strange is, that the `tests/shell` does.

Can anyone confirm, or is it just me?

PeterKietzmann, 2015-07-15:
I tested the `ng_networking`  and `shell` example on current master with an iot-lab_M3 node and was not able to enter commands to the shell.

kaspar030, 2015-07-15:
I git the same issue. I think it is newlib-platforms using uart0. Working on it...

kaspar030, 2015-07-15:
#3415 should fix it for now.

BytesGalore, 2015-07-15:
nice thx. closed via #3415

PR #3415 (2015-07-15): sys: fix uart0
#3161 broke newlib platforms when module_uart0 was used.

This PR bypasses uart0 to uart_stdio when the latter is linked in.

fixes #3413, tested on native, msba2, iot-lab_M3, samr21-xpro.

(uart0 must die!)
closed 2015-07-15 by BytesGalore