RIOT-OS/RIOT #3417: features lost when removing obsolete network stack

kaspar030, 2015-07-15:
This issue is supposed to track features that will be removed by #3334.
Feel free to update.

(empty for now)

authmillenon, 2015-07-15:
@cgundogan might shed some light on the status of `ng_rpl`, @Lotterleben how far is AODV on that front, @benpicco and @OlegHahm what about OLSR?

OlegHahm, 2015-07-15:
OLSR was never merged anyway.

OlegHahm, 2015-07-15:
@kaspar030, what about ccn-lite?

authmillenon, 2015-07-15:
And OpenWSN?

OlegHahm, 2015-07-15:
@thomaseichinger is adapting it to the new driver anyway. Hopefully finished this week.

Lotterleben, 2015-07-15:
@authmillenon  @BytesGalore and I have started to port AODVv2 yesterday (and have merged the the PR integrating the FIB today)... I have no idea how smoothly this will go, but I'll keep you posted.

haukepetersen, 2015-07-15:
I think TCP is under construction, right @brummer-simon?

brummer-simon, 2015-07-20:
Yes it is.

authmillenon, 2015-08-11:
Added NHDP as it uses `socket_base` (which has no source files now).

OlegHahm, 2015-12-08:
A shiny new CCN-Lite version has just been merged.

authmillenon, 2015-12-08:
\o/

kaspar030, 2016-07-30:
Does anyone still care about this?

OlegHahm, 2016-07-30:
I think the only thing that was part of the old stack that is still missing, is TCP support, right? And AFAIK this is hopefully ready soon. Hence, Fine with closing this one.

kaspar030, 2016-07-30:
Yup.

closed 2016-07-30 by kaspar030