RIOT-OS/RIOT #3418: (Temporarily) lost features after radical cleanup

haukepetersen, 2015-07-15:
Here is a list of features, that would be temporarily lost, when doing a radical cleanup by removing all deprecated platforms/drivers and other modules. In the current state we can surely not drop everything in the list (e.g. native, msp430 support)! But this list should give us some guidance on what needs to be done...

Stuff that we get rid of for good:
- sys/uart0
- sys/chardev
- all calls to `dINT`/`eINT` (already done?)

Features that need adjustments (and are candidates to be dropped). Checked items are already being worked on.

CPUs:
 - [x] cpu/arm7_common
 - [x] cpu/arm7
 - [ ] cpu/cc430
 - [x] cpu/lpc2387
 - [ ] cpu/mc1322x
 - [ ] cpu/msp430-common
 - [ ] cpu/msp430fxyz
 - [ ] cpu/native (uart peripheral driver is missing)
 - [ ] cpu/x86

Boards:
 - [ ] boards/native
 - [ ] boards/x86-multiboot-common
 - [ ] boards/msb-430
 - [ ] boards/pttu
 - [ ] boards/msb-430-common
 - [ ] boards/qemu-i386
 - [ ] boards/z1
 - [x] boards/avsextrem
 - [ ] boards/msb-430h
 - [ ] boards/telosb
 - [x] boards/msba2
 - [ ] boards/redbee-econotag
 - [ ] boards/chronos
 - [ ] boards/msba2-common
 - [ ] boards/wsn430-common
 - [ ] boards/wsn430-v1_3b
 - [ ] boards/wsn430-v1_4

Drivers:
 - [ ] drivers/sht11 (also usable with the `dht` driver?)
 - [ ] drivers/smb380
 - [ ] drivers/cc110x_legacy
 - [ ] drivers/cc110x_legacy_csma
 - [ ] drivers/cc2420
 - [ ] drivers/lm75a        
 - [ ] drivers/ltc4150

Others:
 - ?

haukepetersen, 2015-07-15:
Please edit/extend this list!

authmillenon, 2015-07-15:
Kindof duplicate of (or overlap with) #1540

haukepetersen, 2015-07-15:
True. Actually I was looking for #1540 earlier, but could not find it... How about closing #1540 and keep this issue, as this is more detailed and more up-to-date?

OlegHahm, 2015-09-03:
updated the list

kaspar030, 2015-09-05:
updated

cgundogan, 2015-09-10:
added a word about `desvirt`. 0e4386a46fc82e606d466964fcc6c41cd5bc1186 kicked out the `-t` parameter from the native executable that specifies a TCP port for the stdout redirection.

OlegHahm, 2017-01-15:
This looks actually doable for the release.

haukepetersen, 2017-11-02:
It seems to me, that everything we didn't (re-)merge until today can't be important. So closing this issue now.

closed 2017-11-02 by haukepetersen