RIOT-OS/RIOT #3422: NDP: broken when running with 6lowpan IPHC enabled

gebart, 2015-07-15:
I'm guessing it is because the destination multicast address gets compressed in a weird way (ff02::1:ffab:4012 becomes ff02::ab:4012) on the sender or handled wrongly on the receiving end if this is expected behaviour.

See https://www.cloudshark.org/captures/2a764c64eb9d

Dump was taken over the air using a 802.15.4 link layer sniffer. Communication is between two mulle nodes running examples/ng_networking on #3419

gebart, 2015-07-15:
Found the problem. https://github.com/RIOT-OS/RIOT/blob/master/sys/net/network_layer/ng_sixlowpan/iphc/ng_sixlowpan_iphc.c#L576 should be 48 bits version

authmillenon, 2015-07-15:
Can you maybe open a fixing PR please? Will review it.

gebart, 2015-07-15:
I fixed it for the multicast but the subsequent ping packets are completely corrupt (destination address compression field is set to RESERVED according to wireshark)

authmillenon, 2015-07-15:
That might be because there is some mix-up of byte-order. I tried to unify that with #2885 once but lost track.

authmillenon, 2015-07-15:
Are you receiving the packets regardless or is decoding correct?

authmillenon, 2015-07-15:
> I fixed it for the multicast but the subsequent ping packets are completely corrupt (destination address compression field is set to RESERVED according to wireshark)

Wait, what do you mean by "REVERSED"?

gebart, 2015-07-15:
my sniffer is receiving the packets, but the other node ignores them (which it should, they're broken).

gebart, 2015-07-15:
not "reversed", "Reserved"
it seems like the multicast bit gets set on the outgoing ping unicast packets. See the cloudshark dump at https://www.cloudshark.org/captures/1f94c42b9fe9

authmillenon, 2015-07-15:
Aaahhh, yes and the statefull compression should not be used either (unless of course you filled the context buffer with something ;-))

gebart, 2015-07-15:
The stateful flag is set on some packets but not all. Seems like a corrupt pointer or something being used?

gebart, 2015-07-15:
sorry, the other way around, the multicast flag is set on some packets, the stateful flag seems to be set all the time

authmillenon, 2015-07-15:
Found it (I think). I'm opening a PR right now

authmillenon, 2015-07-15:
There was an indexing error.

authmillenon, 2015-07-15:
#3424

authmillenon, 2015-07-15:
I guess we can close this one then, right?

closed 2015-07-15 by authmillenon

gebart, 2015-07-15:
Yes, thank you for your swift bug finding skills!

authmillenon, 2015-07-15:
I would like to be them swifter to find that damn race condition bug in the gnrc stack :(