RIOT-OS/RIOT #3433: ng_at86rf2xx: set_state is messy

gebart, 2015-07-16:
I can't follow the code at https://github.com/RIOT-OS/RIOT/blob/master/drivers/ng_at86rf2xx/ng_at86rf2xx_getset.c#L349

Why does the driver go to `TRX_OFF` regardless of the state parameter and after that only switch states if the new state is one of `TX_ARET_ON` or `RX_AACK_ON`?

I'm trying to figure out why it seems like the driver doesn't receive some packets sent over the air from another node when I am testing ng_rpl.

gebart, 2015-07-16:
another Q: Is it possible to read the status register if the transceiver is in SLEEP mode? the data sheet for 212b says no, so I wouldn't really expect the if (state == SLEEP) bit to work either...

thomaseichinger, 2015-07-16:
* ad `TRX_OFF`: looking at the "Extended Operating Mode State Diagram" Figure 7-10 212b datasheet I read it you can't change from `TX_ARET_ON` to `RX_AACK_ON` directly and vice versa. While it would arguably make more sense to change to `PLL_ON` instead to save state transition time.

* ad SLEEP mode: Agreed, I think it we should check sleep pin instead.

gebart, 2015-07-16:
I will prepare a PR

gebart, 2015-07-16:
The `TX_ARET_ON` -> `TRX_OFF` -> `RX_AACK_ON` transition may be what is making my nodes miss the replies from Contiki node (the reply coming too quickly)

thomaseichinger, 2015-07-16:
`TRX_OFF` -> `PLL_ON` transition takes ~160us.

thomaseichinger, 2015-07-16:
The data sheet conceals however transition times between `TX_ARET_ON` <-> `PLL_ON` <-> `RX_AACK_ON`.

gebart, 2015-07-16:
hmm, my problems could also be the transceiver waiting for ACKs and ACKs disabled on the Contiki side.

thomaseichinger, 2015-07-16:
`TX_ARET_ON` should only wait for ACK when the ACK requested bit is set in the header of the sent packet.

gebart, 2015-07-16:
The radio completely misses the replies from my Contiki node. On the other hand, when the Contiki node sends NS the RIOT node replies with NA packets, so it seems to work one way.

gebart, 2015-07-16:
I turned on auto ACK in Contiki, and Wireshark agrees that ACKs are being sent properly.

 - RIOT to RIOT pings work fine
 - RIOT to Contiki ping replies are sent according to the sniffer but not seen by RIOT.
 - RIOT to RIOT ping roundtrip is >20 ms (22-24 typ.)
 - RIOT to Contiki ping roundtrip is <13 ms (11-12 typ.) [1]

Could it be that the Contiki node responds before the RIOT node has had time to reconfigure the radio?

[1]: The roundtrip could be even shorter since the timestamps are taken by a Python script on my PC, not by the sniffer and the connection between the sniffer and the PC is a 115200 baud UART.

gebart, 2015-07-16:
Turning off ENABLE_DEBUG makes the RIOT node see the replies from Contiki, so there is a timing problem.

thomaseichinger, 2015-07-16:
> * RIOT to Contiki ping replies are sent according to the sniffer but not seen by RIOT.

"...not seen by RIOT" at which point?

gebart, 2015-07-16:
RIOT sends echo request to Contiki. Contiki responds with echo reply. RIOT does not register the echo reply packet, but my sniffer sees it. That is with ENABLE_DEBUG (1) in ng_at86rf2xx_netdev.c. With ENABLE_DEBUG (0) it works. So I guess I caused that error by trying to analyse other problems.

gebart, 2015-08-17:
fixed by #3434

closed 2015-08-17 by gebart