RIOT-OS/RIOT #3438: shell: input hangs on iot-lab_m3

authmillenon, 2015-07-18:
49ea7a3 seems to break input on iot-lab_m3. After talking to @OlegHahm and @kaspar030 I know that they don't seem to see this bug, so I'm posting my toolchain version, too:

```
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/bin/../lib/gcc/arm-none-eabi/4.9.3/lto-wrapper
Target: arm-none-eabi
Configured with: /build/buildd/gcc-arm-none-eabi-4.9.3.2015q1/src/gcc/configure --target=arm-none-eabi --prefix=/build/buildd/gcc-arm-none-eabi-4.9.3.2015q1/install-native --libexecdir=/build/buildd/gcc-arm-none-eabi-4.9.3.2015q1/install-native/lib --infodir=/build/buildd/gcc-arm-none-eabi-4.9.3.2015q1/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/build/buildd/gcc-arm-none-eabi-4.9.3.2015q1/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/build/buildd/gcc-arm-none-eabi-4.9.3.2015q1/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/build/buildd/gcc-arm-none-eabi-4.9.3.2015q1/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/build/buildd/gcc-arm-none-eabi-4.9.3.2015q1/install-native/arm-none-eabi --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors' --with-multilib-list=armv6-m,armv7-m,armv7e-m,cortex-m7,armv7-r
Thread model: single
gcc version 4.9.3 20150303 (release) [ARM/embedded-4_9-branch revision 221220] (GNU Tools for ARM Embedded Processors) 
```

I have this bug on testbed too, so it does not seem to be openocd

authmillenon, 2015-07-18:
(Reverting the commit works for me, not using uart0 does not).

OlegHahm, 2015-07-18:
I can confirm that it's working for me on current master (8850eb4035fed92ae414fd6801ac70898fc7444b) locally and on the testbed.

Toolchain is:
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.8.4 20140526 (release) [ARM/embedded-4_8-branch revision 211358]

authmillenon, 2015-07-18:
Debugging this a little deeper revealed that the UART IRQ handler only handles the first 2 characters and is never called again afterwards.

authmillenon, 2015-07-18:
Typing one character long commands seems to work indefinitely.

authmillenon, 2015-07-18:
@haukepetersen is observing the bug too and on it already.

thomaseichinger, 2015-07-18:
I can confirm this bug happening (sometimes, with some compiled binaries commands are handled correctly, in some not (without changes to UART/syscalls/shell of course)).

```bash
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/local/Cellar/arm-none-eabi-embedded-gcc/20140805/bin/../lib/gcc/arm-none-eabi/4.8.4/lto-wrapper
Target: arm-none-eabi
Configured with: /Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/src/gcc/configure --target=arm-none-eabi --prefix=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/install-native --libexecdir=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/install-native/lib --infodir=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/install-native/arm-none-eabi --build=x86_64-apple-darwin10 --host=x86_64-apple-darwin10 --with-gmp=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/build-native/host-libs/usr --with-mpfr=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/build-native/host-libs/usr --with-mpc=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/build-native/host-libs/usr --with-isl=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/build-native/host-libs/usr --with-cloog=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/build-native/host-libs/usr --with-libelf=/Users/build/GCC-4-8-build/gcc-arm-none-eabi-4_8-2014q3-20140805/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-lstdc++ -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors' --with-multilib-list=armv6-m,armv7-m,armv7e-m,armv7-r
Thread model: single
gcc version 4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision 213147] (GNU Tools for ARM Embedded Processors)
```

cgundogan, 2015-08-17:
@authmillenon does the shell works properly for you now?

authmillenon, 2015-08-17:
Yes. Can be closed.

closed 2015-08-17 by authmillenon