RIOT-OS/RIOT #3451: extra branch on RIOT-OS/RIOT

gebart, 2015-07-20:
Accidental push to wrong repo?
https://github.com/RIOT-OS/RIOT/tree/ngnetworking_readme

x3ro, 2015-07-20:
Yup, getting #3444 merged and then deleting the branch asap :)

gebart, 2015-07-20:
Maintainers: Please do not use the SSH URL for the organisation repository, using the HTTPS link instead makes these kinds of mistakes less likely.

authmillenon, 2015-07-20:
from #3449 (due to me being stupid): 

With the `git://` URL it's even lesser likely. HTTPS still allows Basic Auth (which can be stored), git protocol does not support push. Also, it's [slightly faster](https://git-scm.com/book/be/v2/Git-Internals-Transfer-Protocols).

x3ro, 2015-07-20:
@gebart @authmillenon HTTP works unreliably on the HAW network, though (sometimes pull/pushes are completely stalled until ^C). `git://` is blocked completely (yes, we have very capable admins).

kaspar030, 2015-07-20:
> HTTP works unreliably on the HAW network

I hope that's only with github?

x3ro, 2015-07-21:
Yes, that's only for Git over HTTP :D

cgundogan, 2015-07-21:
I close this since #3444 got merged. Further discussions regarding best practices on how to push branches or how to do branch management can be taken to the mailing list or a separate issue with proper naming

closed 2015-07-21 by cgundogan