RIOT-OS/RIOT #3461: _pktbuf_add_unsafe incorrectly adjusts data pointer

smee204, 2015-07-20:
I think in ng_pktbuf.c _pktbuf_add_unsafe incorrectly adjusts the pkt->data pointer. I find that the address is invalid after this function has completed.

Currently it does this:
pkt->data = (void *)(((uint8_t *)pkt->data) + size);

I think it should do this:
pkt->data = (void *)(((uint8_t *)pkt->data) + (size*4));

authmillenon, 2015-07-20:
What value does `size4` have?

gebart, 2015-07-20:
(edited to add code formatting)

gebart, 2015-07-20:
I don't understand why you need `size * 4`

smee204, 2015-07-20:
Sorry yes now I look at it again the code was correct. I will close this issue.

I think the problem is with my compiler/hardware settings. I must have it setup so pointers have to be 4 byte aligned...

closed 2015-07-20 by smee204