RIOT-OS/RIOT #3469: astyle: Update docs

gebart, 2015-07-21:
According to @authmillenon it should no longer be necessary to run `keyword_space.sed` any more

authmillenon, 2015-07-21:
I did not test that, just conjectured.

A-Paul, 2015-07-21:
Have been curious about that.

    1  s/\bif(/if (/
    2  s/\bswitch(/switch (/
    3  s/\bcase(/case (/
    4  s/\bwhile(/while (/
    5  s/\bfor(/for (/
    6  s/\bdo(/do (/
    7  s/! /!/

*Lines 1,2,3,4,5* are covered by used astyle parameter `--pad-header`. *Line 6* isn't, as you wouldn't have that in C likes.  
There seems to be nothing in *astyle* which could perform the substitution in *Line 7*.

gebart, 2015-07-21:
@A-Paul I don't know what you mean with _6_, `do` is a keyword in C.

A-Paul, 2015-07-21:
@gebart, yes it is. But followed by *left curly bracket* and not by *left parenthesis*. I should have been more specific.

gebart, 2015-07-21:
Ah, that's a typo. Good job spotting it!

OlegHahm, 2016-03-09:
obsolete

closed 2016-03-09 by OlegHahm