RIOT-OS/RIOT #3475: ng_at86rf2xx: Receive interrupt handler is not called anymore when hammered

authmillenon, 2015-07-22:
When I ping in `examples/ng_networking` with very short intervals (eg `ping6 1000 <ip addr> 10 10`) on both `iot-lab_M3` and `samr21-xpro`, I get only ping timeouts after a while and the neighbor cache of the pinged node is empty. When I reset the sender I get ping timeouts immediately, when I reset the receiver the pings work again.

To debug I put a `puts("r")` in the interrupt handler:

```diff
diff --git a/drivers/ng_at86rf2xx/ng_at86rf2xx.c b/drivers/ng_at86rf2xx/ng_at86rf2xx.c
index 5d5a6a0..2f60a0f 100644
--- a/drivers/ng_at86rf2xx/ng_at86rf2xx.c
+++ b/drivers/ng_at86rf2xx/ng_at86rf2xx.c
@@ -44,6 +44,7 @@ static void _irq_handler(void *arg)
     msg_t msg;
     ng_at86rf2xx_t *dev = (ng_at86rf2xx_t *) arg;
 
+    puts("r");
     /* tell driver thread about the interrupt */
     msg.type = NG_NETDEV_MSG_TYPE_EVENT;
     msg_send(&msg, dev->mac_pid);
```

and apparently the handler is never called again on the receiver when the ping timeouts happen, though I see neighbor solicitations fly over the air.

OlegHahm, 2015-07-22:
> I see neighbor solicitations fly over the air.

https://www.youtube.com/watch?v=CFto_NgmFGE

thomaseichinger, 2015-07-22:
https://github.com/RIOT-OS/RIOT/pull/3434 might be of interest.

gebart, 2015-07-22:
These problems may have been caused by the driver getting stuck on TX mode after two transmissions back to back replacing the idle state member variable.

PeterKietzmann, 2015-07-23:
I ran a "hammering-ping-test" on two samr21-xpros 
```
ping6 100000 <ip addr> 10 0
```
with this result:
```
2015-07-23 09:27:55,367 - INFO # 100000 packets transmitted, 99958 received, 1% packet loss, time 2349.06906416 s
```
Not too bad :-) . I think #3434 has done us really good. Can we close this issue, what do you think?

authmillenon, 2015-07-23:
Yapp.

authmillenon, 2015-07-23:
(ran 3 parallel experiments on IoT-Lab yesterday. All 90 minutes with 3 different setups:
* 1 node to 1 node without IPHC
* 1 node to 1 node with IPHC
* 2 nodes to 1 node with IPHC

All with 10 ms delay between pings. Was successful)

closed 2015-07-23 by authmillenon