RIOT-OS/RIOT #3502: log: use logging instead of DEBUG where appropriate for warnings and errors

OlegHahm, 2015-07-27:
Some error messages and warnings in gnrc (and probably elsewhere) are only printed if DEBUG for this module is enabled. While it is okay to discard error messages for production code, it would make testing and debugging a lot more easy if these warnings and errors would always show up, if compiled with `DEVELHELP`. Using `LOG_WARNING` or `LOG_ERROR` in these cases would be the better choice.

OlegHahm, 2015-07-27:
Should we also replace other uses of `DEBUG` with `LOG_DEBUG` (or whatever log level seems to appropriate)?

authmillenon, 2015-07-27:
Would be more unified, I think.

OlegHahm, 2015-07-27:
Disadvantage is that logging cannot be dis- or enabled per file anymore.

PeterKietzmann, 2015-07-27:
Even if I would prefer to have a "_uniform_" logging style, for me it is important to enable debug-messages just in one file.

kaspar030, 2015-07-27:
We had that discussion back when the new logging API was introduced.

AFAIR, the logging API should be used for printing anything that might give a clue on how the system behaves as a whole (cross-module), e.g., ```LOG_DEBUG("at86rf212: received packet of length 120b\n")```, and ```DEBUG()``` is per-file for actually debugging that module while developing it, e.g., ```DEBUG("at86rf212: STATE=0x4711\n")```.

Maybe ```LOG_DEBUG``` should be renamed to ```LOG_VERBOSE``` to point out the intended usage?

We could also drop ```DEBUG``` and make ```ENABLE_DEBUG``` force ```LOG_DEBUG```-output per file.

authmillenon, 2015-07-27:
> We could also drop DEBUG and make ENABLE_DEBUG force LOG_DEBUG-output per file.

I thought this is already the case oO.

OlegHahm, 2015-07-27:
> We could also drop DEBUG and make ENABLE_DEBUG force LOG_DEBUG-output per file.

I would vote for this option. Actually, something like
```C
#undef LOG_LEVEL
#define LOG_LEVEL LOG_DEBUG
```
should enable per file debugging, too.

PeterKietzmann, 2015-07-28:
I don't know the outcomes of the logging API discussions, but I think @OlegHahm's proposal is comfortable

OlegHahm, 2015-07-28:
I tested with #3506 and it seems to work as expected.

kaspar030, 2015-07-28:
I''d rather go for
```#ifdef ENABLE_DEBUG 1
#undef LOG_LEVEL
#define LOG_LEVEL LOG_DEBUG
#endif```
in log.h,
dropping debug.h completely,
and putting ```#define ENABLE_DEBUG (0)``` in front of ```#include "log.h"```.

OlegHahm, 2015-07-28:
Fine with that, too.

OlegHahm, 2015-07-28:
Apparently @kaspar030 disagrees (see #3506).

However, I think we should not hide helpful error messages within `DEBUG`. Hence, I would either vote for overhauling #3506 and revert some `LOG_ERROR/_WARNING` calls to `LOG_DEBUG`/`DEBUG` or replace them with an `assert`.

Option 1 will create some overhead if building with `LOG_LEVEL > LOG_LEVEL_NONE`, option 2 will lead to a system exit whenever such a condition is hit.

OlegHahm, 2015-07-28:
@kaspar030, looking at, for instance, zep:
In [zep_init](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/application_layer/ng_zep/ng_zep.c#L94) I completely that (LOG_)DEBUG is sufficient. I would even argue that don't need the different return values.

But in [_send](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/application_layer/ng_zep/ng_zep.c#L208) I'm not so sure, what would be the best option. The return value is not evaluated. Hence, errors might lead to strange behavior. We could either insert assertions here, which might unnecessary stop the execution for a non-critical event or we could go for `LOG_ERROR` adding some code in non-production code to reveal the problem.

authmillenon, 2015-07-28:
The return values are not evaluated for the most part, but they might come in handy when adapting for #3362.

gorbondiga, 2019-03-28:
Hi, 
as I am new in here and I would like to get involved with this riot-os I think this could be a good ticket for me to start with, could I take it?

kaspar030, 2019-03-28:
@gorbondiga Welcome! :wink:

Depending on how familiar you are with the C preprocessor, this might be quite challenging.

Just take a look at "core/include/log.h" and "code/include/debug.h" and the hackery used there to e.g., not actually print if the stack size is too small. If we'd replace all DEBUG() calls with LOG_DEBUG(), we'd need to replicate that.

miri64, 2020-04-01:
Anyone willing to address this?

closed 2020-11-05 by stale[bot]