RIOT-OS/RIOT #3510: cpu: cortexm_common: set pendSV to lowest priority

OlegHahm, 2015-07-28:
In #3450 the pendSV IRQ was set to the default priority in order to work around a mostly inexplainable issue with the USART ISR. Once this problem has been understood, this might be reverted.

kaspar030, 2015-07-29:
So, Cortex-M MCU's *do* preempt a currently active exception when a higher-priotity (lower priority value) exception occurs. That means if pendSV has a lower priority (higher priority value), it will be interrupted by any ISR's.

As pendSV calls ```sched_run```, that function might be interrupted by an ISR. But in our current design, any ISR could call funtions manipulating the scheduler's state, e.g., by sending a message or locking/unlocking a mutex. This can cause the scheduler to get it's state changed while running, which is is not designed for.

The actual issue we were facing showed behaviour that could result in unprotected access to thread states:
A thread tried to lock a mutex, which "at the same time" got unlocked, but in a racey fashion so that the unlocking thread never got rescheduled.

One way to circumvent this is to disable nested interrupts (and pendSV was the only ISR with a lower priority, so the only exception a Cortex would interrupt), or to disable other interrupts before calling ```sched_run```.

Disabling/reenabling interrupts would add two instructions per context switch, disabling nesting would add one additional scheduler run per ISR that gets queued while another ISR is active, if that ISR would ask for a context switch.

In english, with a *very* high rate of context-switch-causing ISRs, disabling ISRs is more effective cycle-count wise, if that is only a rare case (less than one every ~(nr of cycles per context switch) context switches), giving pendSV default priority is more effective.

So, IMHO until we see *any* need of improved interrupt latency or *any* sign of performance problems caused by running our scheduler too often, I think we are perfectly fine with pendSV running at default priority.

@gebart @haukepetersen ?

OlegHahm, 2015-08-02:
Makes absolute sense to me. AFAIR not having nested interrupts was always an (unwritten) precondition for any work on the kernel/scheduler.

closed 2015-08-02 by OlegHahm

gebart, 2015-08-03:
@kaspar030 :+1: Your dissection of the problem gives a better view of the issue and after reading it I agree that pendSV should have the same priority as the other IRQs.