RIOT-OS/RIOT #3546: USB Device Stack

jfischer-phytec-iot, 2015-08-03:
For some time now I have been working on the USB Device Stack for RIOT. Currently I have a functioning device stack with CDC ACM device support (loopback).  It is progressing slowly, because I do it only in my free time, but steadily. I plan to open a PR at end of the week.
The stack shall/will be designed to run on built-in USB-Controllers (on ARM SoCs or similar). 

Here I would like to put my ideas on the stack to discussion and track the progress.

I see three steps as the main task:

1.  Implementation of USB Device Stack with CDC-ACM support, thus RIOT can act (inside a USB adapter) as a border router.

2. Implementation of USB HCI protocol, thus RIOT can be used inside a USB BLE adapter (if BLE LL is finished).

3. Since there is no common interface such as HCl for IEEE802.15.4 devices under Linux, it is close to take it from bluez and/or adapt it to use with Linux IEEE802.15.4 subsystem.

I think if it is so far, RIOT will be a universal OS for IoT. Especially the step three may increase the popularity of RIOT and increases the acceptance of IEEE802.15.4 devices.

The TODO list for the stack would look like this:

* [ ] USB Device Stack with basic functionality (EP0 only).
  - [ ] CDC-ACM support

* [ ] Implementation of USB HCI protocol
  - [ ] ...

* [ ] Adaption of HCI interface for Linux IEEE802.15.4 subsystem
  - [ ] RIOT-OS as Hard-MAC ...
  - [ ] ...

kaspar030, 2015-08-03:
Nice!

gebart, 2015-08-03:
Thank you for taking care of such a large task in your free time!

PeterKietzmann, 2015-08-03:
Wow :-) !

sgso, 2015-08-03:
I'm looking forward to use/test this for the rf231usb.

OlegHahm, 2015-08-03:
Awesome!

jfischer-phytec-iot, 2015-08-05:
We need a PID / VID for the stack (maybe more PIDs). Are Openmoko PIDs is a possibility? Can someone please requested a PID for RIOT?

http://wiki.openmoko.org/wiki/USB_Product_IDs

gebart, 2015-08-06:
@jfischer-phytec-iot good idea!
Do we need more than one Product ID to distinguish different applications (or use cases)?
It would be useful to be able to distinguish between for example SLIP interfaces and debug message TTY ports, but that can also be done via one of the string properties (iSerialNumber, iManufacturer, iProduct).

miri64, 2016-10-17:
Fixes https://github.com/RIOT-OS/RIOT/issues/2284

PeterKietzmann, 2018-06-26:
Unfortunately this project has been scrubbed. Closing issue (for now)

closed 2018-06-26 by PeterKietzmann

danpetry, 2018-06-26:
Shame, this looks cool

aabadie, 2018-06-26:
There is a starting work made by @astralien3000 and based on this original PR #3890. This feature is very useful indeed.