RIOT-OS/RIOT #3548: PLL samr21: text different from settings

bapclenet, 2015-08-03:
The message in periph_conf [[1]](https://github.com/RIOT-OS/RIOT/blob/master/boards/samr21-xpro/include/periph_conf.h#L57) for the samr21 says :
`` The actual core frequency is adjusted as follows: ``
`` * ``
``  * CORECLOCK = 8MHz / DIV ``

And then 
`` #define CLOCK_USE_PLL       (1) ``

Should we follow the text and set ``CLOCK_USE_PLL`` to 0 or change the text? I think we already had this discussion earlier

PeterKietzmann, 2015-08-03:
Did someone already "play around" with and without the PLL? If the *"cost of a less stable clock"* is really present, I' vote for adapting the doc. Otherwise I tend to use the potentially more power efficient solution.

bapclenet, 2015-08-04:
@PeterKietzmann, I used it to review the PR and it worked but I don't use now since 8MHz is enough for me.
> Otherwise I tend to use the potentially more power efficient solution

I Agree!

PeterKietzmann, 2015-08-04:
@haukepetersen, @thomaseichinger any experiences?

haukepetersen, 2015-08-06:
Yapp, I wrote that part :-)

I actually don't see anything wrong in the text: if you choose the option without PLL, the formula you state is correct. If you use the PLL, the paragraph above that describes the clock calculation for that case.

I would strongly opt against disabling the PLL for a simple reason: wihtout PLL we restrict the main clock to 8MHz. While this might be enough for some cases, the default setting for ALL of our boards is to run with full clock speed. The notion is, that you have all the performance and flexibility by default and that you can then go ahead and tweak for more low-power whenever you have a specific use-case and application...

PeterKietzmann, 2015-08-06:
Well, that is IMO a reasonable answer (this issue is labelled as question). Are we d'accord with that? I am. If @bapclenet is also, I think we can close this issue. Otherwise: @bapclenet I think your proposed solution to adapt the text is it. If you feel like improving the doc, just open a small PR. But IMO the difference gets clear when reading the whole text

bapclenet, 2015-08-06:
For the full clock speed, yes we should follow the default setting. Concerning the text, I read it as "By default, 8MHz is chosen" but actually it means "If you choose The internal Oscillator, the actual core frequency is adjusted as follows ..."

Fine for me!

closed 2015-08-06 by bapclenet