RIOT-OS/RIOT #3574: sys: net: fib is using deprecated net_help

kaspar030, 2015-08-06:
While removing the legacy network stack, I decided not to remove net_help as fib is still depending on it.

A look at the code shows that fib is mainly using net_help's pton function, which could be replaced by gnrc's ipv*_str_to_addr() functions. Also, some defines are taken from net_help.

BytesGalore, 2015-08-06:
I will change that, so net_help can be finally removed.

cgundogan, 2015-08-06:
But wouldn't this also make the gnrc independent fib gnrc dependent?

kaspar030, 2015-08-06:
> But wouldn't this also make the gnrc independent fib gnrc dependent?

only until we factored out the string-to-address conversion functions.
Maybe we should stick with net-help until we've done that?

BytesGalore, 2015-08-06:
I will try to remove every dependency to the old- and gnrc stack.
(soon, I'm at work now)

authmillenon, 2015-09-21:
Should be fixed, since #3878 removed `net_help` and FIB is still compiling.

closed 2015-09-21 by authmillenon