RIOT-OS/RIOT #3588: Receiving of fragmented 6LoWPAN UDP packets does not work

tfar, 2015-08-08:
See pcap here, send by Linux: https://www.cloudshark.org/captures/980bc3eb4583

Which produces log messages (6lo and ipv6) as follows:
```
2015-08-07 23:34:00,448 - INFO # > 6lo: NG_NETDEV_MSG_TYPE_RCV received
2015-08-07 23:34:00,451 - INFO # 6lo: received 6LoWPAN fragment
2015-08-07 23:34:00,453 - INFO # 6lo rbuf: 1st fragment.
2015-08-07 23:34:00,456 - INFO # 6lo: waiting for incoming message.
2015-08-07 23:34:05,454 - INFO # 6lo: NG_NETDEV_MSG_TYPE_RCV received
2015-08-07 23:34:05,457 - INFO # 6lo: waiting for incoming message.
2015-08-07 23:34:05,460 - INFO # ipv6: NG_NETAPI_MSG_TYPE_RCV received
2015-08-07 23:34:05,470 - INFO # ipv6: Received (src = fe80::c07d:1441:7d0d:9fef, dst = fe80::5855:5f70:108b:c07e, next header = 58, length = 40)
2015-08-07 23:34:05,473 - INFO # ipv6: handle ICMPv6 packet (nh = 58)
2015-08-07 23:34:05,476 - INFO # ipv6: forward nh = 58 to other threads
2015-08-07 23:34:05,482 - INFO # ipv6: unable to forward packet as no one is interested in it
2015-08-07 23:34:05,485 - INFO # ipv6: waiting for incoming message.
2015-08-07 23:34:05,488 - INFO # ipv6: NG_NETAPI_MSG_TYPE_SND received
2015-08-07 23:34:05,493 - INFO # ipv6: set payload length to zu (network byteorder 0018)
2015-08-07 23:34:05,496 - INFO # ipv6: set next header to 58
2015-08-07 23:34:05,500 - INFO # ipv6: set packet source to fe80::5855:5f70:108b:c07e
2015-08-07 23:34:05,504 - INFO # ipv6: calculate checksum for upper header.
2015-08-07 23:34:05,507 - INFO # ipv6: removed old interface header
2015-08-07 23:34:05,510 - INFO # ipv6: add interface header to packet
2015-08-07 23:34:05,513 - INFO # ipv6: send unicast over interface 7
2015-08-07 23:34:05,516 - INFO # ipv6: send to 6LoWPAN instead
2015-08-07 23:34:05,519 - INFO # 6lo: NG_NETDEV_MSG_TYPE_SND received
2015-08-07 23:34:05,523 - INFO # 6lo: iface->max_frag_size = 102 for interface 7
2015-08-07 23:34:05,527 - INFO # 6lo: Send SND command for 0x200025d4 to 7
```

It's current master, only added the ```6lo rbuf: 1st fragment``` line to see what code is executed.
The packet is never passed to my UDP server thread listening for it. For packets build from a single frame, they are passed to the UDP server thread.

OlegHahm, 2015-08-08:
Can you specify your setup, please?

Btw: I recently noticed that fragmented packets towards a Linux host weren't processed properly either.

tfar, 2015-08-08:
My setup is like https://github.com/RIOT-OS/RIOT/wiki/How-to-install-6LoWPAN-Linux-Kernel-on-Raspberry-Pi#example-gateway-configuration-using-ulas-unique-local-addresses ( notice the ```rmmod nhc_udp```, to disable UDP header compression on linux, as RIOT does not like that).

In addition to that I have a c++ program that opens a UDP socket and sends the UDP packet mentioned in the pcap to the RIOT host.

Let me know if you need more info.

authmillenon, 2015-08-09:
Since I don't own the hardware to test this, I reassign @OlegHahm for now.

authmillenon, 2015-08-09:
I know however, what the problem is: For some reason the parameters to the 6LoWPAN fragmentation headers in your dump are the values (datagram size and offset) *without* compression, while our implementation (and this was tested again and again with staring at wireshark dumps and is correct as far as my understanding of RFC 6282 goes and IIRC correctly was tested with Contiki) expects the values *with* compression.

tfar, 2015-08-09:
@authmillenon After discussing this issue shortly with @alexaring, one of the developers of the 6LoWPAN stack on Linux, and we're not completely clear to what values and compressions you're exactly referring to. https://tools.ietf.org/html/rfc6282#section-2 states 
```
Section 5.3 of
   [RFC4944] defines the fragment header's datagram_size and
   datagram_offset values as the size and offset of the IPv6 datagram
   before compression.  As a result, all fragment payload outside the
   first fragment must carry their respective portions of the IPv6
   datagram before compression.  This document does not change that
   requirement."
```

Are you referring to the datagram_size mentioned there?

authmillenon, 2015-08-09:
Yes, and I know of this section. But regardless, the current implementation is what worked in previous interop-tests and with the wireshark dissector, so I'm quite confused about it.

OlegHahm, 2015-08-09:
@authmillenon, if you like I can give you access to a preconfigured and working raspberry Pi with a IEEE 802.15.4 interface and global IPv6 address.

authmillenon, 2015-08-09:
Okay. Will do that tomorrow :-)

OlegHahm, 2015-08-10:
Interestingly enough the other way around seems also to be not quite stable. After pinging a Linux host with a (potentially broken) fragmented echo request, Linux got a kernel panic. See https://gist.github.com/OlegHahm/efd1c68b3ca9b4b6f653

OlegHahm, 2015-08-10:
The last things I saw in tshark on the Pi were
```
  0.000000 fe80::5855:686c:2535:65b2 -> ff02::1:ff06:cb3 ICMPv6 96 Neighbor Solicitation for fe80::c934:b5fa:5a06:cb3 from 5a:55:68:6c:25:35:65:b2
  1.000098 fe80::5855:686c:2535:65b2 -> ff02::1:ff06:cb3 ICMPv6 96 Neighbor Solicitation for fe80::c934:b5fa:5a06:cb3 from 5a:55:68:6c:25:35:65:b2
  2.000509 fe80::5855:686c:2535:65b2 -> ff02::1:ff06:cb3 ICMPv6 96 Neighbor Solicitation for fe80::c934:b5fa:5a06:cb3 from 5a:55:68:6c:25:35:65:b2
 17.492308 fe80::5855:686c:2535:65b2 -> ff02::1:ff06:cb3e ICMPv6 96 Neighbor Solicitation for fe80::c934:b5fa:5a06:cb3e from 5a:55:68:6c:25:35:65:b2
 17.492835 fe80::c934:b5fa:5a06:cb3e -> fe80::5855:686c:2535:65b2 ICMPv6 96 Neighbor Advertisement fe80::c934:b5fa:5a06:cb3e (sol, ovr) from cb:34:b5:fa:5a:06:cb:3e
 17.544012 fe80::5855:686c:2535:65b2 -> fe80::c934:b5fa:5a06:cb3e ICMPv6 164 Echo (ping) request id=0x0054, seq=1
 17.544501 fe80::c934:b5fa:5a06:cb3e -> fe80::5855:686c:2535:65b2 ICMPv6 164 Echo (ping) reply id=0x0054, seq=1
 22.499594 fe80::c934:b5fa:5a06:cb3e -> fe80::5855:686c:2535:65b2 ICMPv6 96 Neighbor Solicitation for fe80::5855:686c:2535:65b2 from cb:34:b5:fa:5a:06:cb:3e
 22.527445 fe80::5855:686c:2535:65b2 -> fe80::c934:b5fa:5a06:cb3e ICMPv6 80 Neighbor Advertisement fe80::5855:686c:2535:65b2 (sol)
 23.754347 fe80::5855:686c:2535:65b2 -> fe80::c934:b5fa:5a06:cb3e ICMPv6 164 Echo (ping) request id=0x0055, seq=1
 23.754762 fe80::c934:b5fa:5a06:cb3e -> fe80::5855:686c:2535:65b2 ICMPv6 164 Echo (ping) reply id=0x0055, seq=1
```

alexaring, 2015-08-10:
Probably the error handling is broken at lowpan_xmit_fragment. I saw this issue earlier, I preparing patches for this. Should be fixed by:
https://gist.github.com/alexaring/f55c09304aa5144773af

authmillenon, 2015-08-10:
(was prompted by a bug of mine were an empty fragment (with payload 0) was send after all fragments were send by me).

alexaring, 2015-08-10:
Don't know, but the stacktrace smells more like an issue in transmit handling not receive handling.
Does this happen when you transmit a empty fragment after all fragments from RIOT to Linux?

authmillenon, 2015-08-10:
Yes.

alexaring, 2015-08-10:
mhh, okay. Then thanks for this note. I will keep that in my mind, currently there are pending patches which makes the receive handling "more robust", maybe it will fix this issue. At first I will fix the error handling in lowpan_xmit_fragment because the stacktrace looks like that something is wrong there and I currently see this issue right now. And yes, it could happen that you can reach kernel-panics when sending weird 6LoWPAN data. See my suggestions at:
http://www.spinics.net/lists/linux-wpan/msg02588.html
but we are on it to make it all "robust". We have this mainline because some historical reasons. ;-)

tfar, 2015-08-11:
Any news on this? Is there a branch I can test that'll join fragmented 6LoWPAN UDP packets?

authmillenon, 2015-08-11:
I'm working on it. Yesterday I managed to get sending straight, but did not manage to test receiving. Have a look at https://github.com/authmillenon/RIOT/tree/ng_sixlowpan/fix/issue-3588. I will continue working on it either tonight or tomorrow, as time allows.

tfar, 2015-08-11:
Thanks, I appreciate it. I've tested your branch but the UDP messages did not read my UDP server event loop. Will test a new version as soon as an update is out.

OlegHahm, 2015-08-11:
> Since I don't own the hardware to test this, I reassign @OlegHahm for now.'

The situation reverted. ;)

authmillenon, 2015-08-11:
> Thanks, I appreciate it. I've tested your branch but the UDP messages did not read my UDP server event loop. Will test a new version as soon as an update is out.

As I've said: receiving is in a state of "I need to finish this before I go home". So there might be a lot of bugs in there still. ;-)

authmillenon, 2015-08-11:
@tfar see #3613.

PR #3613 (2015-08-17): ng_sixlowpan: fix #3588
Fixes #3588.
commit 1129f567 (2015-08-17): ng_sixlowpan: fix #3588
commit bbe6ec50 (2015-08-17): Merge pull request #3613 from authmillenon/ng_sixlowpan/fix/issue-3588

ng_sixlowpan: fix #3588
closed 2015-08-17 by OlegHahm